SpenceKonde/megaTinyCore #1045: Attiny402 Pin Change Interrupt

bbhatt1080, 2024-01-05:
Hello,

Is there an example of the PCI and the timer interrupt? for the ATTINY402 on the Arduino platform? 
I tried a couple of 3rd party libraries - which gave compile errors.

I am using the megaTinyCore

Thank you

=B. Bhatt=

hmeijdam, 2024-02-09:
Should you still need it. Works on 402 as well.

```
// ATtiny412 / ARDUINO Pin Change Interrupt Example
//                                    _____
//                             VDD   1|*    |20  GND
//           (DAC)   (AIN6) PA6  0   2|     |19  4~  PA3 (AIN3)(SCK)(EXTCLK)
//                   (AIN7) PA7  1   3|     |18  5   PA0 (nRESET/UPDI)
// (MOSI)(TXD*)(SDA) (AIN1) PA1  2   4|_____|17  3   PA2 (AIN2)(MISO)(RXD*)(SCL)
//
//

#define LED1 PIN_PA6
#define LED2 PIN_PA7

// when PA3 is touched to ground, the pin change ISR will trigger and turn on LED1

void setup() {
pinModeFast(LED1, OUTPUT);
pinModeFast(LED2, OUTPUT);
  //  PORTA_PIN3CTRL = PORT_PULLUPEN_bm | PORT_ISC_LEVEL_gc; // interrupt active at low level
  PORTA_PIN3CTRL = PORT_PULLUPEN_bm | PORT_ISC_FALLING_gc; // falling edge
  //  PORTA_PIN3CTRL = PORT_PULLUPEN_bm | PORT_ISC_RISING_gc; // rising edge
  //  PORTA_PIN3CTRL = PORT_PULLUPEN_bm | PORT_ISC_BOTHEDGES_gc; // both edges
}

void loop() {
    digitalWriteFast(LED2, HIGH);
    delay(1000);
    digitalWriteFast(LED2, LOW); 
    digitalWriteFast(LED1, LOW);
    delay(1000);
}

ISR (PORTA_PORT_vect) {
    digitalWriteFast(LED1, HIGH);
  byte flags = VPORTA.INTFLAGS;  // read the interrupt flags
  // writing "1" to the interruptflag bit will clear it.
  VPORTA_INTFLAGS = flags; //clear interrupt flags
}
```

SpenceKonde, 2024-03-28:
Closing because not a defect in the core. Please also see the interrupt documentation. I discuss this at considerable length.

closed 2024-03-28 by SpenceKonde