SpenceKonde/megaTinyCore #1049: UPDI Iinitialization failed on attiny1604

charbelB7, 2024-01-12:
I'm using an attiny1604 devboard from soldered.com (https://soldered.com/product/dasduino-lite/) and i tried to program it with serial updi  but as said in the title i get the following error message :
Traceback (most recent call last):
`File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 286, in <module>
    main()
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 128, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 201, in pymcuprog_basic
    args_start)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\pymcuprog_main.py", line 545, in _start_session
    backend.start_session(sessionconfig)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\backend.py", line 362, in start_session
    sessionconfig.interface_speed)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\programmer.py", line 83, in setup_device
    options=self.options)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\nvm.py", line 42, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\nvmserialupdi.py", line 54, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\serialupdi\application.py", line 79, in __init__
    datalink.init_datalink()
  File "C:\Users\delta\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\serialupdi\link.py", line 44, in init_datalink
    raise PymcuprogError("UPDI initialisation failed")
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed
Failed programming: uploading error: exit status 1`
I am using the following CH340 adapter :  https://www.seeedstudio.com/USB-To-Uart-5V-3V3-p-1832.html Which has no internal resistor for TX.
the circuit i am doing is the following :  

[image]
with a schottky diode and a 470 ohm resistor in serial with the UPDI pin.  I am programming it with 230400 bauds and and even tried to reconfigure the serial port's uart settings as using even parity bits and 2 stop bits but to no success. Any idea on how to fix this?

charbelB7, 2024-03-13:
could you please answer this problem, its been two months and still doesn't work

hmeijdam, 2024-03-13:
Are you sure this is a board without a bootloader that has disabled UPDI and turned it into a Reset pin? I see your board has already a CH340G and USB connector 
And the RESET pin is shared with UPDI and has a whole circiutry connected to it, that may interfere with UPDI programming.

[image]

You need to have a fresh Attiny1604 without additional components soldered to the UPDI pin, to make this work. I don't think that is what you have.

SpenceKonde, 2024-03-30:
For UPDI to work with that board you need to
1. Ensure that JP1 (autoreset) is not connected
2. Ensure that you connect the wires in the correct order. The pin order there is **bad**; it violates basic principles of applied murphology`*` and sets a trap for customers. The correct order is to have Gnd in the middle, so that even if you get the connector backwards, you're swapping Vcc and UPDI, which is unlikely to cause hardware damage. That board puts UPDI in the middle, so that if you get the connector backwards, Vcc and Gnd are reversed. This is invariably fatal to the chip, and often lets out smoke. 

That whole board appears to have made a large number of poor design decisions.

But that updi header, jeez. I guess that's one way to ensure repeat business!

`*` Applied Murphology: Consideration of murphy's law and it's ramifications as an integral part of the design process. If anything can go wrong, it will; Over a large enough sample everything that can go wrong will, most many times. Therefor consideration should always be given to "How likely is it for this thing to go wrong? Plugging a connector in backwards when it's not polarized is very easy, and so if its often connected or disconnected, you assume it will be connected backwards routinely for indefinite lengths of time. In this case we would consider that there are three pin orderings (each of which can be reversed):
* Vcc, UPDI, Gnd (what they did - plugged in backwards, will burn out chip in seconds from reverse polarity.)
* Updi, Vcc, Gnd (also bad - The Vcc and Gnd rail of target would be at Vcc (which is fine except that...) UPDI is connected to Gnd. Since both Gnd and Vcc on the target are approximately Vcc, that would mean that the external voltage applied to an I/O pin was less than Gnd from the chip's perspective, so that UPDI pin would be aced if it was wired like that
* Vcc, Gnd, UPDI (good - chip will sometimes power on depending on load, but will never be damaged...

fritzlb, 2024-04-21:
From the picture of your serial adapter it seems like that thing has got LEDs for RX/TX. As per documentation, in case the RX LED just shorts the RX line over a resistor to VCC, that won’t work with UPDI.

SpenceKonde, 2024-12-24:
I'm closing this, you absolutely cannot have an LED anywhere between the RX pin of serial adapter and the UPDI pin of target if you're programming a tiny. I don't think that this is the case when programming an AVR Dx or Ex, however - the difference is that their UPDI pin (PF7) is not the same pin that the HV pulse goes on, so it's a full strength I/O pin. The pin that takes the HV pulse (which is also the reset pin, PA0 on tinyAVR and PF6 on Dx/Ex) has extremely weak pin drivers where it has pin drivers at all (the tinyAVRs and classic parts. All Dx/Ex, they admitted that the reset-as-IO was such a shit output that they might as well just not bother with the HV-tolerant pin drivers, and only let you set it input. 

*The pin drivers on PA0 are so weak that you could drive an LED with it without a resistor, and be disappointed that the LED was too dim.* These are the same pin drivers used for UPDI communication, so you have to be careful to not load the pin even a little bit. If you want a UPDI RX led, you'd need to use a buffer or comparator, or a serial adapter (eg, FT232RL) which has dedicated pins to drive the activity lights.
You *can* put a light on the TX pin of the serial adapter (before the diode or resistor), as that's getting the full drive strength of the serial adapter. 

Finally, note that an RX light is not that helpful on a UPDI programmer - assumiung we have solved the pin drive strength, it's still less than useful:
TX will be on whenever the programmer is sending data. RX would be on whenever the programmer is sending data or the target is sending

A proper TX and RX led for UPDI programming would need an IC or two, because, actual transmit is ongoing when the TX line is low (and the RX line will also be low). Receiving is ongoing when the RX line is low but the TX line is NOT low, and there is a serious communication problem or damage to the programmer if the RX line is high and the TX line is low 
(in all cases I'm talking about the lines measured at the programmer end, before all added circuitry)

.     | TX  | RX   | Idle   | Error state
TX | L      | L     | H      | L
RX | H     | L     | H      | H 

**THAT ALL SAID** the reason that serial adapter isn't working does not appear to be directly driven RX LED, but rather an bungled attempt to avoid this sort of problem, driving it with a transistor. But a transistor still conducts non-negligible current, and UPDI on a tinyAVR has virtually no wiggle room. 

And if they'd used a buffer to drive the LED, it absolutely would work. 

And they didn't need the extra circuitry on the TX led. It's a CH340, it's got pin drivers nearly as strong as an AVR. You can totally put a TX led directly on the TX line from the serial adapter at least for most serial adapters. That's what I do, and yes the serial adapters I made have had endless problems (I'm at rev E of one design, Rev B and Rev A of another design, and dropped yet another design at Rev B and another after getting the first boards back (I will never again break my vow not to buy from FTDI again. That was clearly god punishing me for breaking a vow).

closed 2024-12-24 by SpenceKonde