SpenceKonde/megaTinyCore #1059: drazzy.com ssl cert expired ?

br74649, 2024-02-16:
Hi, 

Im just trying to download the core via boards manager and getting an error in arduino ide 

"Error downloading http://drazzy.com/package_drazzy.com_index.json"

when i hit it with a browser its being redirected to https and chrome complains about the cert which appears to have expired today 

[image]

Any chance this can be renewed? 

Thanks

Tony

HeadBoffin, 2024-02-16:
Hopefully the LetsEncrypt can be renewed asap and that Spence is A-OK

Mialee85, 2024-02-17:
Yeah, the certificate for dazzy.com ran out yesterday. Hope somebody is gonna pick that up quick.

HeadBoffin, 2024-02-17:
Two potential solutions:

Use your browser to go to https://drazzy.com/package_drazzy.com_index.json and when it complains about the certificate expiring, tell it that it's OK, save the file to disk and then in your IDE you can use file:///path/from/root/of/disk/to/file/package_drazzy.com_index.json or for Windows file:///c:/path/from/root/of/disk/to/file.package_drazzy.com_index.json.

Or if you figure that I've got many better things to do than try to hack in to peoples computers because they visited one of my web servers, you can use https://descartes.net/package_drazzy.com_index.json as a temporary measure. I can't hack in to your computer by altering the JSON file either but I could do odd things to the board support files if I had nothing better to do.

Anyone who wants to get all pointy fingered will have me delete this workaround, so please just don't.

br74649, 2024-02-18:
thanks @HeadBoffin , i did do the file workaround - would rather it working as the dr intended , hope he is ok

handy workaround for now though 

Thanks

Tony

cygig, 2024-02-28:
Oh dear, I couldn't even download anything today on my new laptop. I could still access it a week ago. His website is down as well. Is he ok?

Zeron, 2024-02-29:
Seems to be working now.

HeadBoffin, 2024-03-01:
It is indeed updated - the server may have been restarted or something else pressed.

However there have been no updates since Oct '23 so I think we still need to track down Spence to check he's OK

nickzam, 2024-03-18:
Hi, I tried the [aforementioned fixes](https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469) for Arduino IDE 1.8.5, 1.8.13 (all under Windows 10 Home) installed via installer (from scratch) and standalone, with and without deletion of  preferences.txt file to trigger new libraries update. 

Search term "megatinycore" just not appearing in the list of libraries, there are other that appears (Attiny_PWM, Attiny_Slow_PWM etc), but not MegaTinyCore.

Any ideas how to install MegaTinyCore into Arduino IDE manually or use UPDI on Arduino as a programmer on PlatformIO for Attiny 402 ?

nickzam, 2024-03-18:
@per1234 Thank you for pushing me to right direction! I've added http://drazzy.com/package_drazzy.com_index.json into "Additional Boards Manager URLs" add it finally appeared in **Boards Manager** :)

SpenceKonde, 2024-03-30:
I am still here, and I have been periodically getting enough time to respond to a small number of issues. But 

Those fucking certs are a constant headache. The automatic renewal methods that they talk about DO NOT WORK. 
The only time they get updated is when I manually log into the machines. The duration that the certs are valid for is long enough for me to forget the steps to manually renew. For reasons I don't understand, cron jobs don't run. I have no idea why, I set them up, see that they're in the crontab and then check back a few days later when it should have run and see no evidence that anything happened when the cronjob should have run. I don't have enough leverage on the guy who knows linux right now to get him to help out, I think he wishes I knew someone else who was competent at linux.

And I have a bunch of other irons to fire; I have to collate 6 years of tax documents - the bloodsuckers want to know crazy numbers. Revenue!? Expenses?! How the fuck should I know? Money come money go! Meanwhile time seems to be going by faster and faster, like I have less and less time to do more and more things. I do not know how anyone could ever be bored.

closed 2024-03-30 by SpenceKonde

hmeijdam, 2024-03-30:
Nice to see you back here.
My grandpa often said. "you have only 2 certainties in life, Death and Taxes".  
So the "good" news is, you have not been hit by the first one ;-)

srw777, 2024-03-30:
I would be happy to help you administer the server.  I have been running
Linux webservers since 1996 (my first one was connected via 128kbps ISDN).
I have received so much value from your cores, but really don't have the
skills to contribute to the code... would be elated to help in a way that I
am able.  Also, I completely agree with the LetsEncrypt renewal thing...
I've fought with that for years.  I kind of enjoy reading your rants when
they come. :-)

M2tec, 2024-07-06:
> Two potential solutions:
>
> Use your browser to go to https://drazzy.com/package_drazzy.com_index.json and when it complains about the certificate expiring, tell it that it's OK, save the file to disk [...]

Had a similar ssl cert issue with the Micronucleus packages used in the json file. Replacing https with http resolved the issue for me.

vadpol, 2024-07-09:
> Had a similar ssl cert issue with the Micronucleus packages used in the json file. Replacing https with http resolved the issue for me.

Could you please be more specific on this: yes, I could download and edit the json file, but can't save it back to repository of course. How do you use this edited json file with http iso https to get right reference to Micronucleus packages? I can't install the Attiny Core because of this Micronucleus download error message.

M2tec, 2024-07-09:
Search for these instances in the json file. 
              "url": "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip"
and replace with:
              "url": "http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip"

vadpol, 2024-07-09:
> Search for these instances in the json file. "url": "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip" and replace with: "url": "http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip"

That's clear, thanks! But how to substitute this file while IDE is taking the existing wrong one from url where it is not changed?

brandock, 2025-03-01:
Is this related to the certificate expiration and is there a workaround to get the bootloader uploader in the right place within Arduino IDE? (I am using 1.8.19 on a silicon Mac).

I used the local file work-around to get the board manager working. While trying to burn a bootloader after installing the board manager, I get the following.
```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
	at processing.app.Editor.lambda$handleBurnBootloader$61(Editor.java:2385)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.
Error downloading http://drazzy.com/package_drazzy.com_index.json
```

I hate SSL certificate renewals deeply.