SpenceKonde/megaTinyCore #1060: EVSYS gen1 not declared?

FactoryOptic, 2024-02-17:
'''
#include <Event.h>
#include <Logic.h>

void setup() {

    // EVSYS.ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN6_gc; /* Asynchronous Event from Pin PA6 */
    Event0.set_generator( gen0::pin_pa6 ); // Set pin Pa6 as event generator, Event0

    // EVSYS.ASYNCCH1 = EVSYS_ASYNCCH1_PORTB_PIN6_gc; /* Asynchronous Event from Pin PB6 */
    Event1.set_generator( gen1::pin_pb6 ); // Set pin PB6 as event generator, Event1

  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
'''

Getting

'''

/private/var/folders/lc/c27z1qfj4j97xx6m63x10yjh0000gn/T/.arduinoIDE-unsaved2024116-16182-1tekyd1.szz7/sketch_feb16a/sketch_feb16a.ino:11:27: error: 'gen1' has not been declared
     Event1.set_generator( gen1::pin_pb6 ); // Set pin PB6 as event generator, Event1
                           ^~~~

Using library Event at version 1.3.0 in folder: /Users/someone/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/libraries/Event 
Using library Logic at version 1.3.1 in folder: /Users/someone/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/libraries/Logic 
exit status 1

Compilation error: 'gen1' has not been declared
'''

SpenceKonde, 2024-03-28:
Improper tests were carried out to see whether that namespace should exist, and if so, make it available.

SpenceKonde, 2024-03-28:
Fixed by dc86020

closed 2024-03-28 by SpenceKonde