SpenceKonde/megaTinyCore #1068: MegaTiny core not listed in board manager

KevinTetreault, 2024-03-03:
When I add the url http://drazzy.com/package_drazzy.com_index.json to preferences and click OK, a message with a progress bar appears indicating that the json file is being loaded. But afterwards, the megaTiny core is not present in the board manager list. I'm using Arduino IDE Version 2.2.1.

KevinTetreault, 2024-03-04:
I've installed the core on an old laptop with Arduino IDE Version 1.8.19. All works well, so it is an issue with regards to the new IDE. Any plans to support the new 2.x IDE ?

mechatroniks-git, 2024-03-05:
It was taken down - see  https://github.com/SpenceKonde/ATTinyCore/issues/847

SpenceKonde, 2024-03-30:
> Any plans to support the new 2.x IDE ?

Yes, the plans were fully implemented as of late summer 2023; since then we've been waiting for a working version of Arduino 2.x to be released. The bug is known - Arduino has had a fix since september, but have chosen not to do a normal release of it uh... because... I don't know. 

I can't change the filename (which unlike most packages, adhered to the official guidelines; that was what triggered the bug; The scofflaws who ignored the guidelines were unimpacted, but I did what they said and included the full domain name. Like their document showed, with the .com. Then when they were "fixing" what wasn't broke, they somehow wound up breaking handling of filenames with more than 1 `.` in them. People who ignored the guidelines (apparently everyone else) didn't include the domain name in the filename because that would break every existing installation in the field because apparently that's what they index installed cores by (that's why they gave us a formulaic way to generate one that nobody used - that was how they would prevent unintentional name collisions (no effort appears to have been made other than verifying the SSL cert, and I was given the impression that so much as thinking about renaming that file would cause the dead to rise again to destroy the living, turning the sky blood red and reducing cities to ruins - in addition to causing problems for all users who had my cores installed currently - so that approach is a non-starter, I don't have time for the inquiries that come as it is, I live in a city and it being reduced to ruins would be an unacceptable imposition on my quality of life, whatever blotted out the shorter wavelengths of sunlight would reduce the efficiency of solar energy, which our species can ill afford right now. So it seems that I don't have a clear right way to fix this, and I never expected we would still be talking about this issue now.

closed 2024-03-30 by SpenceKonde