SpenceKonde/megaTinyCore #1070: Bug Report: Serial.write() Skips Bytes in Transmission

Sandvoxel, 2024-03-05:
**Title:** Bytes Skipped During Serial Transmission with ATtiny3227 on Custom Board

**Environment:**
- **Hardware:** Custom board using ATtiny3227
- **Software Version:** Arduino IDE 2.3.2
- **Operating System:** Windows 11

**Description:**
I've encountered an issue where `Serial.write()` appears to skip some bytes during transmission on a custom board utilizing an ATtiny3227 microcontroller. This issue was observed when transmitting a specific set of bytes via serial communication at a baud rate of 115200. Despite the expectation that all bytes are sent sequentially without omission, some bytes seem to be missing from the output.

**Steps to Reproduce:**
1. Initialize the serial communication with `Serial.begin(115200);`.
2. Transmit a predefined byte array using `Serial.write()` inside the `setup()` function without any loop.
3. Observe the transmitted bytes on the serial monitor or through a serial terminal program.

**Expected Result:**
All bytes in the array are transmitted correctly and can be seen on the receiving end in their entirety.

**Actual Result:**
Some bytes appear to be skipped in the transmitted sequence. The expected output was 
`[1, 1, 17, 77, 83, 53, 54, 49, 49, 32, 102, 111, 117, 110, 100, 46, 10, 211, 91, 0]`
but the observed output was 
`[1, 1, 77, 83, 53, 54, 49, 49, 32, 102, 111, 117, 110, 100, 46, 10, 211, 91, 0]`
indicating that the byte `17 (0x11)` is missing from the sequence.

**Example Code:**
```cpp
void setup() {
    Serial.begin(115200);

    uint8_t data[] = {0x01, 0x01, 0x11, 0x4D, 0x53, 0x35, 0x36, 0x31, 0x31, 0x20, 0x66, 0x6F, 0x75, 0x6E, 0x64, 0x2E, 0x0A, 0xD3, 0x5B};
    for (unsigned char i : data) {
        Serial.write(i);
    }
    Serial.write(0);
}

void loop() {}
```
**Additional Information:**
- Adjusting the baud rate did not resolve the issue.

hmeijdam, 2024-03-05:
Check with a better terminal emulator.
When I run your sketch and print it to HTerm I get the characters exactly as you describe

[image]

Sandvoxel, 2024-03-05:
My mistake this was caused by the ftdi chip being in Xon Xoff mode causing the missing bytes

closed 2024-03-05 by Sandvoxel