SpenceKonde/megaTinyCore #1076: Upload sketch with SerialUPDI HV Programmer doesn't work

avandalen, 2024-03-23:
Hi,
I'm working on a new description of how to burn the bootloader:
https://avdweb.nl/arduino/attiny3217/bootloader-attiny3217

I have a problem with the megaTinyCore version 2.6.10 and the SerialUPDI_HV_Programmer

The used SerialUPDI HV Programmer is this:
https://github.com/wagiminator/AVR-Programmer/tree/master/SerialUPDI_HV_Programmer

This works well: I can upload sketches with the SerialUPDI HV programmer, with the high voltage pulse disabled.

Problem:
Uploading sketches with the 12V pulse enabled, gives an error message

(In the year 2020, I had successfully worked with the megaTinyCore and I was able to upload sketches via UPDI and also via the burned bootloader.)

Arduino settings:
[image]

Scope
[image]

Error messages
[without HV pulse OK.txt](https://github.com/wagiminator/AVR-Programmer/files/14731976/without.HV.pulse.OK.txt)

[with HV12V pulse NOT OK.txt](https://github.com/wagiminator/AVR-Programmer/files/14731977/with.HV12V.pulse.NOT.OK.txt)

pcfreak1201, 2024-03-25:
May be the timing is too tight between pulse and prog?
https://onlinedocs.microchip.com/oxy/GUID-A2109DC3-B5FF-4E1B-BDB5-622C21D35F43-en-US-5/GUID-93D88655-0A9C-4923-A03F-077491438F6F.html

avandalen, 2024-03-25:
Maybe, but I cant do anything about it, it is controlled by the megaTinyCore and the Arduino IDE.

closed 2024-03-26 by avandalen

avandalen, 2024-03-26:
I closed this issue because the actual problem was this:
https://github.com/SpenceKonde/AVR-Guidance/issues/20