SpenceKonde/megaTinyCore #1079: ATtiny3216 SPI master to ESP8266 SPI slave

Hai-Elec, 2024-04-04:
Hi,
I can read and send data with ATtiny3216 (SPI master) to ESP8266 on ESP12F module (SPI slave).
But I do not understand why some first characters of string is bypass.
With MPLABXIDE it by pass 2 first characters.
With Arduino, it by pass 7 first characters. 
With Arduino example SPI coding, at first try, it can show first character, but after that, I do not understand why it by pass. Now it never give me the first character anymore.
I test with setting clock SPI DIV128, I try change internal clock ATtiny3216 from 20MHz to 1MHz, result is the same.
I connect directly 5V SPI  pin to ESP-12F, because, if I connect SS pin of ESP8266 to level shifter IC TXS0108E, it can not boot, try set pin Low by ATtiny 3216 when start up, but it still not boot(I rechecked TXS0108E it still work OK when I control level of SS pin of ESP8266 3V3 by ATtiny3216 5V pin). Recheck information on Arduino forum, I directly connect SPI to ESP8266. 
Now I can send data via SPI OK, but I must add some unnecessary first characters, so, I add here for more information.
Please share after check coding:
Arduino of ATtiny3216:
`

// inslude the SPI library:
#include <SPI.h>
  #define SSPIN  PIN_PA4
  #define MOSI  PIN_PA1
  #define MISO  PIN_PA2
  #define SCK  PIN_PA3
char txMsg[] = "1234567abcdefghijklmnopqrstuvwyz12345!";///it by pass first 7 character

void setup() {

  pinMode(SSPIN, OUTPUT);
  digitalWrite(SSPIN, LOW);
  pinMode(MOSI, OUTPUT);
  pinMode(MISO, INPUT);
  pinMode(SCK, OUTPUT);
  delay(1000);

  digitalWrite(SSPIN, HIGH);

  delay(3000);

  // initialize SPI:
 
  SPI.begin();
  SPI.setClockDivider(SPI_CLOCK_DIV128); 

}

void loop() {
 
  digitalWrite(SSPIN, LOW);

  for(long i=0; i< sizeof(txMsg); i++)
  {
    SPI.transfer(txMsg[i]); 
  }

  digitalWrite(SSPIN, HIGH);

  delay(1000);
 
}
`
For SPI slave of ESP12F, I use example code SPISlave_Test. Just reup here for more convenient. Please check:
`/*
    SPI Slave Demo Sketch
    Connect the SPI Master device to the following pins on the esp8266:

    GPIO    NodeMCU   Name  |   Uno
  ===================================
     15       D8       SS   |   D10
     13       D7      MOSI  |   D11
     12       D6      MISO  |   D12
     14       D5      SCK   |   D13

    Note: If the ESP is booting at a moment when the SPI Master has the Select line HIGH (deselected)
    the ESP8266 WILL FAIL to boot!
    See SPISlave_SafeMaster example for possible workaround

*/

#include "SPISlave.h"

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  // data has been received from the master. Beware that len is always 32
  // and the buffer is autofilled with zeroes if data is less than 32 bytes long
  // It's up to the user to implement protocol for handling data length
  SPISlave.onData([](uint8_t *data, size_t len) {
    String message = String((char *)data);
    (void)len;
    if (message.equals("Hello Slave!")) {
      SPISlave.setData("Hello Master!");
    } else if (message.equals("Are you alive?")) {
      char answer[33];
      sprintf(answer, "Alive for %lu seconds!", millis() / 1000);
      SPISlave.setData(answer);
    } else {
      SPISlave.setData("Say what?");
    }
    Serial.printf("Question: %s\n", (char *)data);
  });

  // The master has read out outgoing data buffer
  // that buffer can be set with SPISlave.setData
  SPISlave.onDataSent([]() {
    Serial.println("Answer Sent");
  });

  // status has been received from the master.
  // The status register is a special register that bot the slave and the master can write to and read from.
  // Can be used to exchange small data or status information
  SPISlave.onStatus([](uint32_t data) {
    Serial.printf("Status: %u\n", data);
    SPISlave.setStatus(millis());  // set next status
  });

  // The master has read the status register
  SPISlave.onStatusSent([]() {
    Serial.println("Status Sent");
  });

  // Setup SPI Slave registers and pins
  SPISlave.begin();

  // Set the status register (if the master reads it, it will read this value)
  SPISlave.setStatus(millis());

  // Sets the data registers. Limited to 32 bytes at a time.
  // SPISlave.setData(uint8_t * data, size_t len); is also available with the same limitation
  SPISlave.setData("Ask me a question!");
  Serial.println("on");
}

void loop() {}
`

SpenceKonde, 2024-12-19:
See hspi_slave.c in the ESP8266 package. the one I'm seeing has on line 29
```c
static uint8_t _hspi_slave_buffer[33];
```
Then down on 59
```c
         _hspi_slave_buffer[32] = 0;
         for(i=0; i<8; i++) {
                data=SPI1W(i);
                _hspi_slave_buffer[i<<2] = data & 0xff;
                _hspi_slave_buffer[(i<<2)+1] = (data >> 8) & 0xff;
                _hspi_slave_buffer[(i<<2)+2] = (data >> 16) & 0xff;
                _hspi_slave_buffer[(i<<2)+3] = (data >> 24) & 0xff;
          }
```

closed 2024-12-19 by SpenceKonde

SpenceKonde, 2024-12-20:
First test: 
V1. Use a storage scope to monitor the data lines, set to decode. Confirm that you can see the tiny transmitting. 

V2. Lower the length of the string to below 32 characters. 

V1 passes if the "missing" characters are seen on the wire
V2 passes if lowering the length of the string below 32 characters makes the problem vanish. 

If A V1/2 test fails, I'm gonna need more information
* The source code running on the tinyAVR (should be easy)

* Less easy, but much more important. I never set up the IDE to program ESP8266's. I have ESPTOOL, but I have sucha backlog atm I can't go learning new architectures. So I need the compiled binary that reproduces the problem (jut like outputting what it gets to the ESP's serial), and the esptool command to upload it - If the above theories don't pan out, this is going to be a bfd to solve. But I think it's just a limit of the ESP8266 library, which you should open as an issue over there - 32 byte serial transfers? Why? Doesn't that thing have kilobtytes of ram?

Reopen if evidence that the problem is in this core.

Hai-Elec, 2024-12-21:
Thank SpenceKonde. Currently, I happy with result after bypass some first character. Generally it still work if I by pass some first character. Currently I send short string with 20 character, so it still not reach to 32 character.
In future, if I need more length, I will back to test. Because I not store the setup File for Saleae logic version 1, so, it can not use to check the signal of SPI on computer screen. Now Saleae logic only have version 2 to download on website. I still not purchase version 2.
The code I use for ATtiny3216SN in first comment, that from the example in Arduino with a little edit to test.
I have just finish a project using ATtiny3216SN send 9 sensor value to ESP8266 via SPI and show value on website and log value to local Xampp server on computer. Just bypass some first character still OK with me.

Hai-Elec, 2024-12-21:
Thank SpenceKonde. Currently, I happy with result after bypass some first character. Generally it still work if I by pass some first character. Currently I send short string with 20 character, so it still not reach to 32 character.
In future, if I need more length, I will back to test. Because I not store the setup File for Saleae logic version 1, so, it can not use to check the signal of SPI on computer screen. Now Saleae logic only have version 2 to download on website. I still not purchase version 2.