SpenceKonde/megaTinyCore #1082: ADCPowerOptions() ADC_ENABLE and ADC_DISABLE appear to be exchanged (re-open #1078)

andrewwatkin, 2024-04-05:
2dad406 does not fix #1078 

The problem is that ADC_ENABLE disables the ADC and ADC_DISABLE enables it.

To make sense, the code in wiring_analog.c around line 419 requires that
```ADC_ENABLE == _ADC_ENABLE_CTRL  | _ADC_ENABLE_VAL```

But this is not the case in Arduino.h around line 224.

``` C
  #define ADC_ENABLE        0x20
  #define ADC_DISABLE       0x30
```
There is also the same problem for the series-0 / series-1 code.

Sorry I am not in a position to retest at the moment - this is just inferred by code inspection.

SpenceKonde, 2024-04-05:
I belive this has been corrected now with 5e3147bb

closed 2024-04-05 by SpenceKonde

andrewwatkin, 2024-04-06:
Looks functionally correct now. Thank you.

 

Wiring_analog.c around line 650 (series 0/1 code path) would benefit from the same changes from numeric constants to _ADC_ENABLE_* #defines as you did in the same file around line 420 (series 2 code path)

 

Wiring_analog.c around line 890 is similar for the ADC1 code path

 

A search in this file for all three occurrences of “temp |= 1; // ADC on” shows the way.