SpenceKonde/megaTinyCore #1093: Two potential solutions:

terrypin999, 2024-04-30:
Two potential solutions:

Use your browser to go to https://drazzy.com/package_drazzy.com_index.json and when it complains about the certificate expiring, tell it that it's OK, save the file to disk and then in your IDE you can use file:///path/from/root/of/disk/to/file/package_drazzy.com_index.json or for Windows file:///c:/path/from/root/of/disk/to/file.package_drazzy.com_index.json.

Or if you figure that I've got many better things to do than try to hack in to peoples computers because they visited one of my web servers, you can use https://descartes.net/package_drazzy.com_index.json as a temporary measure. I can't hack in to your computer by altering the JSON file either but I could do odd things to the board support files if I had nothing better to do.

Anyone who wants to get all pointy fingered will have me delete this workaround, so please just don't.

_Originally posted by @HeadBoffin in https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469_

terrypin999, 2024-04-30:
Hobbyist, not programmer here, and find github navigation hard work. Seemed unable to simply comment to @HeadBoffin 's helpful post, so here instead.
Opened the link you provided. No mention of certificates. How do I get from there to an alternative core, please? Do I simply add
`https://descartes.net/package_drazzy.com_index.json`
to these in IDE Preferences ?
`https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json
http://digistump.com/package_digistump_index.json
http://drazzy.com/package_drazzy.com_index.json
https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
https://raw.githubusercontent.com/MHEtLive/arduino-boards-index/master/package_mhetlive_index.json
`

tyeth, 2024-05-29:
Just to say last time the certificate expired I setup a mirror on a subdomain using cloudflare to proxy the content and provide a valid SSL certificate. This ignores all certificate issues on drazzy.com and is zero cost (free cloudflare plan)

E.g.
https://drazzy.good-enough.cloud [via cloudflare SSL] = https://drazzy.com

So the board support package URL becomes 
https://drazzy.good-enough.cloud/package_drazzy.com_index.json

Here is our board URL change:

https://github.com/adafruit/ci-arduino/commit/6b7a47c49ef963f3712a84ea427a9233bc520cb5

Probably worth me tagging issue #1059

henrygab, 2024-06-12:
Fascinating.  It does seem like a bug in CloudFlare to allow proxy to an HTTPS site where the certificate is expired, though, doesn't it?

tyeth, 2024-06-12:
Hehe, I call it a feature 🐛

I think it's actually part of their ability to protect and very a certificate chain between your server and them (verifying certs from your private certificate signing authority), while offering a public cloudflare certificate for your CDN proxied traffic (web users)

closed 2024-12-14 by SpenceKonde

henrygab, 2024-12-29:
The more I think on this, the more this is a security nightmare.   In the simplest scenario, a MitM attack or rogue server could serve up malicious data, but access through your proxy would hide the fact that the data source has indications of being untrusted.   For those considering this as a solution for themselves, consider the risk you are exposing yourself to (it's your choice in the end).

However, **_please_** don't suggest this as a "good" solution, or enable it broadly for other folks....

tyeth, 2024-12-29:
Welcome to the internet, your consumed SSL certs rarely verify the server
specifically (by IP and unique host fingerprint) so dns hijacking and other
MitM attacks are always a risk. Sadly certificate trust chains are often
compromised too so it's all pretty token gesture security.

Definitely agree though that this obscures the true origin which isn't
great.

If you choose to trust a random persons software (firmware) and hosting
solution then you're in the wonderful place of being both empowered and
considered slightly crazy to some. If you then trust another's proxied
version you're that much further out on a limb (tree branch)...

Also I'm a little surprised it's still a sticking point to get free SSL now
it's basically 2025, maybe providers need to make it automatic.

It's also the case that some devices can't do https and that some shouldnt
have to, but arduino-cli disagrees hence our issues.