SpenceKonde/megaTinyCore #1097: [question]: How to disassemble compiled binaries?

okawo80085, 2024-05-07:
I need to disassemble my compiled binary to check if some optimizations compiled correctly, I tried using the included `avr-objdump`, but it doesn't show `avrxmega3` architecture as supported and `plugin` doesn't work, am i missing something or is that just not supported at the moment?

okawo80085, 2024-05-07:
Supported architectures i see with `avr-objdump`
```
> ~/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objdump -H

~/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objdump: supported targets: elf32-avr elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex
~/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objdump: supported architectures: avr avr:1 avr:2 avr:25 avr:3 avr:31 avr:35 avr:4 avr:5 avr:51 avr:6 avr:100 avr:101 avr:102 avr:103 avr:104 avr:105 avr:106 avr:107 plugin

```

MX682X, 2024-05-15:
The mTC and DxCore usually generate an .lst file that already has the decopmiled code. It is in the same folder as the .elf File (the Path can be found in the compiler logs)

okawo80085, 2024-05-15:
Thats exactly the info that i needed! The flags that i was missing were this (if anyone other than me ever needs it):
```
~/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objdump --disassemble --source --line-numbers --demangle --section=.text <elf file> >  disassembled.lst
```
Thank you for the info @MX682X !

closed 2024-05-15 by okawo80085