SpenceKonde/megaTinyCore #1104: PlatformIO support

Andy79881, 2024-05-25:
I would like to use VScode with platformIO instead of Arduino IDE. Is MegaTinyCore supported? Especially I need to check a certain option. I could not figure out how to do that. Do you have any hints to point me in the right direction? Thanks in advance

T94T, 2024-06-28:
See: https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/PlatformIO.md

You need to set `platform = atmelmegaavr`

SpenceKonde, 2024-07-26:
Yeah, please see that document. And if you have further questions, I would suggest asking a person who has  a high liklihood of knowing the answer (which rules me out)

SpenceKonde, 2025-06-06:
Closing this as not actionable. Please report any issues relating to PIO with the understanding that I can fix a problem if you tell me exactly what is wrong that is causing PIO issues, but I cannot debug, or write documentation or give guidance for PIO users because I don't use PIO and do not have the time to take on learning a new build environment. I hate build environments and

closed 2025-06-06 by SpenceKonde