SpenceKonde/megaTinyCore #1106: Seems like the security certificate for your web site has expired

moose4lord, 2024-06-01:
When I try to install megaTinyCore from the Arduino IDE I get this error:
Some indexes could not be updated. Get "https://drazzy.com/package_drazzy.com_index.json": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: “drazzy.com” certificate is expired: http://drazzy.com/package_drazzy.com_index.json

When I try to access https://drazzy.com/ using Chrome I get this error:

NET::ERR_CERT_DATE_INVALID

This server could not prove that it is drazzy.com; its security certificate expired 4 days ago. This may be caused by a misconfiguration or an attacker intercepting your connection. Your computer's clock is currently set to Saturday, June 1, 2024. Does that look right? If not, you should correct your system's clock and then refresh this page.

hmeijdam, 2024-06-01:
You can try using [this workaround](https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469)

That worked fine last time when the certificate expired.

I believe DrAzzy has some other real life priorities causing him to put his cores on the backburner.

moose4lord, 2024-06-01:
Thanks for the workaround. Worked like a charm.

SpenceKonde, 2024-07-26:
fixed

closed 2024-07-26 by SpenceKonde