SpenceKonde/megaTinyCore #1109: Only first call to analogWrite() working on ATtiny817

caternuson, 2024-06-07:
This may affect other variants, but was discovered and tested using an ATtiny817.
megaTinyCore release 2.6.10

Example sketch to demonstrate issue:
```cpp
// attiny817 PWM pins: 0, 1, 9, 10, 11, 12, 13, 20
// works: 0, 1, 9, 10, 20
// issue: 12, 13

#define PWM_PIN 13

void setup() {
  pinMode(PWM_PIN, OUTPUT);
}

void loop() {
  analogWrite(PWM_PIN, 255);
  delay(2000);
  analogWrite(PWM_PIN, 0);
  delay(2000);
}
```
and an LED attached to the output pin.

For pins 0, 1, 9, 10, and 20, the sketch works as expected. The LED blinks.
For pins 12 and 13, only the first call to `analogWrite()` works. Changing `255` to `127` has the expected effect on brightnesls. But the LED never turns off.

hmeijdam, 2024-06-08:
I can confirm your findings for pin 12 and 13 (PC0 and PC1). With a 3217 it also seems that analogWrite(PWM_PIN, 0) blocks further analogwrite to that pin. Changing it into analogWrite(PWM_PIN,1) gives no problems, so it seems that only setting it to "0" breaks the function.
13 (PC1) can only have PWM from TimerD (WOD), so I suspect that timer is used for analogWrite on these two pins. TimerD is also notoriously difficult to set up,

SpenceKonde, 2024-08-30:
Yes, it's related to a feature designed to improve analogWrite performance. I'm surprised that this has gone unnoticed for this long,...

caternuson, 2024-09-25:
@SpenceKonde Is there an older BSP release version you could recommend as a temporary work around?

Thanks for checking and tagging this issue!

SpenceKonde, 2024-12-18:
Can you please retest with current github? I just fixed several critical PWM bugs.

hmeijdam, 2024-12-18:
I tested with the latest boards manager version and with the Github version on a T416 and see no difference between the two versions. Both don't work as expected. I used PWM pins on "default" setting in the tools menu.
The analogWrite (PWM_PIN, 255) is ignored. When I change 255 into 254 it works fine. When I switch to PIN_PB0 it also works fine.

```
#define PWM_PIN PIN_PC0

void setup() {
  pinMode(PWM_PIN, OUTPUT);
}

void loop() {
  analogWrite(PWM_PIN, 255); // gets skipped
  delay(2000);
  analogWrite(PWM_PIN, 20);
  delay(2000);
  analogWrite(PWM_PIN, 5);
  delay(2000);
}
```

hmeijdam, 2024-12-19:
Seems to work ok now including both endpoint values. Tested with my scope.

SpenceKonde, 2024-12-19:
Yaknow, that "feature" was in for years u_u

closed 2024-12-19 by SpenceKonde

caternuson, 2025-01-03:
@SpenceKonde Sorry for delay. Just got around to testing this. The example sketch from first issue post does work now as expected. An LED attached to either pin 12 or pin 13 blinks.