SpenceKonde/megaTinyCore #1115: ATTINY1616 "UPDI init failed: Can't read CS register" other chips work fine.

jvl88, 2024-06-20:
Trying to upload code to the Adafruit Seesaw ATTINY1616, without success.
Arduino IDE compiler gives me the "UPDI init failed: Can't read CS register" error.

What have I tried so far:
Different programmers
Different cables
Different Arduino IDE versions
Different Chips.

What is strange to me thought is that programming an ATTINY1624 with my UPDI setup works like a charm.
So to me that feels like IDE, Programmer and cables are fine.
It is just the 1616 that throws issues.
Or is there something specific I am missing here? (brain has left the building since it is almost 3 AM here).

Any thoughts on this?
Thank you in advance

grandaspanna, 2024-06-20:
No specific ideas, but I've routinely done 1616 chips with no issues. I source all my chips direct from Microchip or Digikey.

jvl88, 2024-06-20:
Thank you for your response @grandaspanna 
I have 10x 1616's ready for a project, just wanted to do some prototyping (before ordering PCB's) using the 1616 breakout from Adafruit.

Managed to solder some wires to one of the bare chips.
Unfortunately same result, "UPDI init failed: Can't read CS register".
Whilst flashing other chips than 1616, over UPDI works fine. 🤷🏻‍♂️

grandaspanna, 2024-06-20:
That's super weird. Does "burn bootloader" also trigger the same issue? Maybe attach the whole output. Someone smarter than me might pick something up.

FWIW - I'm using a Sparkfun serial/USB adapter with a resistor between RX & TX.

jvl88, 2024-06-20:
It does actually output the same issue.
I use a nanouart with a resistor as well.
But again.. my setup works fine flashing others than the 1616 so I moved away from thinking there is an issue there.

Hope I am posting the output down below correctly, and someone indeed picks up on it :).

```
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port /dev/cu.wchusbserial544C0071381 at 57600 baud.
Target: attiny1616
Set fuses: ['0:0b00000000', '1:0x00', '2:0x02', '5:0b11000101', '6:0x04', '7:0x00', '8:0x00']
Action: erase
pymcuprog.programmer - INFO - Setting up programming session for 'attiny1616'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny1616
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/cu.wchusbserial544C0071381' at '57600' baud
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - WARNING - UPDI init failed: Can't read CS register.
pymcuprog.serialupdi.physical - INFO - Sending double break
pymcuprog.serialupdi.physical - INFO - Double-break sent. Retrying.
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/cu.wchusbserial544C0071381' at '57600' baud
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - WARNING - UPDI init failed: Can't read CS register.
Traceback (most recent call last):
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/prog.py", line 286, in <module>
    main()
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/prog.py", line 128, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/prog.py", line 199, in pymcuprog_basic
    status = pymcu._start_session(backend,
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/pymcuprog_main.py", line 545, in _start_session
    backend.start_session(sessionconfig)
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/backend.py", line 359, in start_session
    self.programmer.setup_device(
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/programmer.py", line 78, in setup_device
    self.device_model = get_nvm_access_provider(self.transport,
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/nvm.py", line 42, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/nvmserialupdi.py", line 54, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/serialupdi/application.py", line 79, in __init__
    datalink.init_datalink()
  File "/Users/----/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.10/tools/libs/pymcuprog/serialupdi/link.py", line 44, in init_datalink
    raise PymcuprogError("UPDI initialisation failed")
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed
pymcuprog.serialupdi.physical - INFO - Closing port '/dev/cu.wchusbserial544C0071381'
Failed to burn bootloader: uploading error: exit status 1
```

grandaspanna, 2024-06-20:
I don't think that's specific to reading the CS register (or any specific register), it's simply not establishing any connection at all. Do you have a scope to see what's happening on the line? Without any prejudice, have you got the right pin on the device?

jvl88, 2024-06-20:
Thanks again @grandaspanna 
UPDI pin is verified and correct (clearly noted on the pcb from Adafruit, but also measured to exclude soldering defect).
I do have a scope yes. Just not sure what to look for.
I got the following signal, around the following lines when trying to flash (as far as I could tell, going back an forth between two displays ;)).
Other than that, signal is pulled up.

```
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
```
[image]

grandaspanna, 2024-06-20:
I'm hoping that might be useful to someone :-). I've just taken delivery of 100 more 1616's, so I hope they behave like my other ones!

hmeijdam, 2024-06-20:
The maximum LOW voltage value on the UPDI pin is defined as 0.3 x VDD (Chapter 36.10 datasheet). It looks like you are feeding off 3.3V, so LOW would need to be < 1V to be registered as such. I can't see it well in your graph, but your LOW look higher on your scope than 1 V.
You could try feeding the Attiny with 5V. Then your LOW would be 1.5V max

jvl88, 2024-06-20:
@hmeijdam Thank you for the suggestion.. Same result though.

jvl88, 2024-06-20:
@hmeijdam scratch my previous message. It actually did do the trick!
Switching from 3v3 to 5v meant switching to a different programmer (that I did not mend). 
Whilst doing so I forgot to add the resistor to that programmer 🤦🏻‍♂️.
My bad..

Still not sure why the 3v3 setup works just fine on other chips than the 1616?
Thank you @grandaspanna & @hmeijdam for helping me out.
Much appreciated.

closed 2024-06-20 by jvl88

hmeijdam, 2024-06-20:
Ok, that's a bummer. This is where I would grab my DIY HV UPDI programmer to see if it's UPDI pin is by accident changed into a GPIO pin.
Now for some of the Attinies (e.g. 412)[ I have been able](https://github.com/SpenceKonde/megaTinyCore/discussions/450) to get the UPDI pin working again by applying a brief 12V pulse with a 12V source. For more recent produced Attinies that no longer works as they require precise timing between the HV pulse and start of UPDI programming. I had to [make another HV Programmer](https://github.com/Dlloydev/jtag2updi/wiki/DIY-HV-UPDI-Programmer#schematic) for those.

hmeijdam, 2024-06-20:
Ok, I see it worked. You may want to flag my first response as solution for future readers. Your scope trace nicely shows the issue.

jvl88, 2024-06-20:
Yes, I wanted to do so. However I can't seem to figure out how?

hmeijdam, 2024-06-20:
> Yes, I wanted to do so. However I can't seem to figure out how? Not getting the option when clicking the "..."

I start to believe you can't in this discussions area, as it needs to be [earmarked as "Q&A" ](https://docs.github.com/en/discussions/managing-discussions-for-your-community/managing-categories-for-discussions#default-categories)by the repository owner to enable flagging solutions.