SpenceKonde/megaTinyCore #1117: Board Manager URL update for Arduino IDE Update Compatibility

DigiMaxIO, 2024-06-24:
In the latest Arduino IDE (2.3.2 @ time of writing), doesn't accept the original URL used in the installation guide. 
However, I got it to work using the raw file from your [ReleaseScripts repo](https://github.com/SpenceKonde/ReleaseScripts/tree/master).

**Error with Old URL:**
[image]

**Old URL:**
~~`http://drazzy.com/package_drazzy.com_index.json`~~ ❌ 

**New Working URL:**
`https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/master/package_drazzy.com_index.json` ✅

ConzZah, 2024-06-30:
Good Evening,
It might be worth noting, that certs for https://azduino.com/bin/micronucleus/ seemingly have also expired, which at first prevented me from installing it.
But long story short, i went and uploaded those to archive and changed the links in "package_drazzy.com_index.json" to: https://archive.org/download/azduino_micronucleus-mirror
respectively, & now it downloads / installs just fine.

modified .json:
https://github.com/ConzZah/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/package_drazzy.com_index.json

or:
https://archive.org/download/azduino_micronucleus-mirror/package_drazzy.com_index.json

(of course it is only a temporary solution, but i hope this can be of help).

Cheers, ConzZah.

SpenceKonde, 2024-07-26:
SSL cert fixed old URLs work again

closed 2024-07-26 by SpenceKonde