SpenceKonde/megaTinyCore #1122: Is there an issue with series-2 chips HV programming?

lsellens, 2024-07-06:
I'm having issues when trying to program an ATtiny424 using a high-voltage (HV) programmer. Programming and setting fuses work perfectly without the HV programmer, but I want to experiment with some flags and need to test that I can recover from it first. I have used the same HV programmer on other chips from series-0 without issue.

However, when I attempt to set fuses or upload a sketch with the HV programmer, I receive the following error:

```
raise PymcuprogError("Unexpected number of bytes in response: "
pymcuprog.pymcuprog_errors.PymcuprogError: Unexpected number of bytes in response: 0 byte(s) expected 1 byte(s)
Failed programming: uploading error: exit status 1
```
This is with arduino ide 2.3.2 megatinycore 2.6.10 on linux

lsellens, 2024-07-06:
I just realized I can read/write fuses, and upload hex files with avrdude using this HV programmer, but can not do any of those things with pymcuprog even with the most recent version in their git on this one chip but it works as expected on other chips from series-0. It's obviously not a megatinycore issue, but I'm going to leave this open for a bit in case someone comes by with a suggestion.

hmeijdam, 2024-07-06:
Been there also. A few months back I had[ this rather lengthy exercise](https://github.com/SpenceKonde/megaTinyCore/issues/1077) to discover that some "older"  generation of the new attinies in the "0" and "1" series accept a simple 12V pulse to set the UPDI pin back to UPDI, Heck I even did it with a [12V coin cell battery](https://github.com/SpenceKonde/megaTinyCore/discussions/450). 
But some of the "1" and "2" series attinies  have implemented a much stricter timing requirement for the 12V pulse. So I built [this HV programmer](https://github.com/Dlloydev/jtag2updi/wiki/DIY-HV-UPDI-Programmer#schematic) and that's the only one I found that works for those Attinies. 
So if you want to disable UPDI pins and turn them into reset pins, but want to be able to go back, You may need to build a second HV programmer.

lsellens, 2024-07-06:
> Been there also. A few months back I had[ this rather lengthy exercise](https://github.com/SpenceKonde/megaTinyCore/issues/1077) to discover that some "older" generation of the new attinies in the [...]

I'm using this one
https://github.com/adafruit/Adafruit-High-Voltage-UPDI-Friend-PCB
strange part is I can use it with avrdude just not pymcuprog not.

hmeijdam, 2024-07-07:
Not strange as your Adafruit programmer is based on Stefan Wagners design for[ this programmer](https://github.com/wagiminator/AVR-Programmer/tree/master/SerialUPDI_HV_Programmer) that just sends a HV pulse.

But it does not generate a powercycle. For the latest generation Attinies you need first a powercycle (PC) and then after about 7 microseconds you need a 12V (HV) pulse. This is called a [PCHV programmer](https://github.com/wagiminator/AVR-Programmer/tree/master/UPDI_HV_Programmer), that he also designed a PCB for.

So I first built the HV programmer on the bottom in this picture, to find out that I also had to make a PCHV programmer (top in the picture). That one based on a Nano is the one that generates both the Powercycle and the 12V pulse. 

[image]

lsellens, 2024-07-07:
It all works fine with avrdude and it doesn't perform any kind of power cycle either?

hmeijdam, 2024-07-07:
They both work via avrdude as that is the underlying programmer tool in the Arduino IDE.
The lower one in the picture does not do power cycle, just a 12V pulse before programming
The upper one provides a power cycle just before the 12V pulse and then the programming follows

SpenceKonde, 2025-05-07:
One very important thing to be aware of: On AVR DD and later, the max voltage on reset is 9.0V ABS MAX, and minimum for HV prog mode is 2v above Vdd, typical given as 7.5, and they repeatedly warn about that.