SpenceKonde/megaTinyCore #1123: Issue with baud rate calculation under certain circumstances

tangiblewaves, 2024-07-07:
I came across this issue:
I used Serial0 (baudrate does not matter here) and found, that the effective baudrate was wrong after Serial0.begin().
Looking into the source of UART.cpp I saw that the baudrate divider is calculated as 
`baud_setting = (((4 * F_CPU) / baud))`
 which is fine for the default value of the Periperal clock (PDIV in CLKCTRL.MCLKCTRLB). But I had changed this; therefore the calculation failed.
Was no issue to workaraund,  just by setting USART.BAUD after Serial0.begin(), but I wonder, if the peripheral clock should be considered in the baudrate calculation?

hmeijdam, 2024-07-07:
Are you using setup() and loop() or just a main()

or can you show a minimal sketch that just demonstrates your problem?

tangiblewaves, 2024-07-07:
I'm using setup() and loop(); good old Arduino style ;.)

Im setup(), I'm calling 
`    _PROTECTED_WRITE(CLKCTRL.MCLKCTRLB, CLKCTRL_PDIV_2X_gc | CLKCTRL_PEN_bm);`

Later in setup() then
`  Serial0.begin(MIDI_BAUDRATE);`

I fix it by calling 
`  USART0.BAUD = ....` after `Serial0.begin`

Really, not a big issue, but in case somebody stumbles upon the same problem...

SpenceKonde, 2024-07-29:
This core follows the paradigm of the stock Arduino core and the vast majority of third party cores when it comes to the matter of users directly accessing hardware registers: We only claim that any API function will do any specific thing as long as the user has not interacted with any hardware register through a means other than an API function. Just like on any other AVR in Arduino-land, if you change the system prescaler, millis and micros will be wrong and baud rates will be wrong. 

This is highly inconvenient, since people may not want to run at the maximum clock speed... **WHICH IS WHY THERE IS AN OPTION IN THE TOOLS SUBMENUS TO SELECT THE CLOCK SPEED**

When the clock speed is set in the tools submenu and is not changed within the sketch itself at runtime, the baud rate is calculated correctly, and timekeeping works. 

When clock speed is changed in any other way, it should not be expected to work. In common with all arduino cores I am aware of, F_CPU is a #defined macro passed to the compiler, and thus is a constant. You would not be happy seeing the impact on code size and performance if F_CPU was not considered constant.

closed 2024-07-29 by SpenceKonde

tangiblewaves, 2024-08-23:
A bit late, but thanks for the clarification!