SpenceKonde/megaTinyCore #1124: Micronucleus certificate has expired, so installation fails

dpharris, 2024-07-08:
Failed to install platform: 'ATTinyCore:1.5.2'. 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: “azduino.com” certificate is expired

This stops the installation.  Is there a workaround?  

Thx

hmeijdam, 2024-07-08:
Yes,, you can find the workaround by looking through the list of existing issues. certificate has expired has been mentioned a few times in the past months.

dpharris, 2024-07-08:
Yes, I did the work around for the expired certificate for the core, but
this is a different expired certificate.

dpharris, 2024-07-08:
I have downloaded the Micronucleus -- where do I put it?

dpharris, 2024-07-08:
Ah --- using "https://archive.org/download/azduino_micronucleus-mirror/package_drazzy.com_index.json" in the Additional boards URL worked the charm.

hmeijdam, 2024-07-08:
Creative workaround :-)

SpenceKonde, 2024-07-26:
Duplicate of issue relating to SSL. Certs renewed.

closed 2024-07-26 by SpenceKonde