SpenceKonde/megaTinyCore #1126: error message when downloading the board JSON file

BroekerFrank, 2024-07-14:
I get an error message when downloading the board JSON file.
This means that the JSON file cannot be loaded.
I am using the Arduino IDE V1.8.19.
This error has been occurring for a few days. I suspect the JSON file is not formatted correctly or the Arduino check is too strict.
Error message:
<Error downloading from http://drazzy.com/package_drazzy.com_index.json>

I have already reported this error some time ago. Unfortunately, the error is now occurring again.

Greetings
Frank

moose4lord, 2024-07-15:
This is an ongoing problem with the security certificate for Mr Konde's web site. See issue #1117 for a workaround.

SpenceKonde, 2024-07-26:
Duplicate closed. I go away for any length of time, and the fucking certs go bad no matter how many instructions for autorenew I set up.

closed 2024-07-26 by SpenceKonde

hmeijdam, 2024-07-26:
> I go away for any length of time.
Good to see you back here.