SpenceKonde/megaTinyCore #1127: Can't install megaTinyCore anymore

avandalen, 2024-07-16:
Hi Spence

Installing the megaTinyCore doesnt work anymore , whats happened?
megaTinyCore is not on the list in board manager

(Note that every time I install another package, I need to install megaTinyCore again to get de ATtiny3217 in the board list)

I always use this:
File -> Preferences> Additional board manager URLs = http://drazzy.com/package_drazzy.com_index.json
Tools -> Boards -> Boards Manager ...
Select "megaTinyCore by Spence Konde" and click "Install". (This can take a while)

hmeijdam, 2024-07-16:
As a workaround you can put this in your: File -> Preferences> Additional board manager URLs

https://archive.org/download/azduino_micronucleus-mirror/package_drazzy.com_index.json

avandalen, 2024-07-16:
Yes, now it works :-)  :-) :-) :-) :-) :-)
I added this to my article here:
https://avdweb.nl/arduino/attiny3217/bootloader-attiny3217

nomakewan, 2024-07-21:
Just as a note, a better URL to use would be this one since it comes from the maintainer directly:

https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/master/package_drazzy.com_index.json

SpenceKonde, 2024-07-26:
The SSL situation is fixed and original links again work. Anyone know how to set up autorenew?

closed 2024-07-26 by SpenceKonde