SpenceKonde/megaTinyCore #1141: Integrate QTouch functionality

Ilidur, 2024-09-12:
Hello, the QTouch functionality for the ATTiny chips is not available in this library. I've manage to put together a functional but slightly hacked together implementation borrowing the work from https://github.com/xukangmin/TinyCore .

My target is the ATTiny1616 so I won't be able to test any suggestions on any other board. Would the project be interested in this contribution? The license of the original library by @xukangmin is MIT.

hmeijdam, 2024-09-12:
The PTC peripheral has always been a mystery to me. A peripheral without any associated registers gives me a headache. 

So is the TinyTouch you refer to really using the PTC peripheral or is it bitbanging touch functionality through ADC, Timer(s) and interrupts?

Ilidur, 2024-09-13:
Yes, it uses the QTouch library, although it is proprietary, comes as a
precompiled library.
https://github.com/xukangmin/TinyCore/tree/master/avr/libraries/Touch/src

MX682X, 2024-09-26:
The code on github has already a PTC library. However, I'm finishing up the DA support right now and made some small changes in the API, like changes to the argument order (it will be x, y instead of y, x for the pin bitmap, should only affect self-cap sensing)

MX682X, 2024-09-28:
The most up-to-date code can be found in the DxCore, no further changes to the API is to be expected. If there won't be any bug complains, it will be copied to this repository aswell.

SpenceKonde, 2024-12-20:
[What MX682X talked about has been merged in :-)](https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/libraries/PTC)

Please go bang on it with a hammer and find some bugs in it, folks :-)

closed 2024-12-20 by SpenceKonde