SpenceKonde/megaTinyCore #1142: SLEEP_MODE_PWR_DOWN have too much consumption

PM04290, 2024-09-13:
Hello,
I use new ATtinys since few weeks (ATtiny84 before, without problem) and i need to have a minimum consumption in sleep mode.
After documention reading, i wrotte the little code below for ATTiny3216 and i have 520µA in sleep mode, too far from AVR datasheet (5µA)
I'am on 5Vdc, 1MHz internal, no BOR, no millis()

I found a few rare posts that talk about the problem but none solve my problem; any idea?

```
// Led ON during 100ms, and power off during 10s
#include "avr/sleep.h"
#define PIN_DEBUG_LED  PIN_PC2
volatile uint16_t sleepCounter = 0;

ISR(RTC_PIT_vect)
{
  RTC.PITINTFLAGS = RTC_PI_bm; // Clear interrupt flag by writing '1' (required)
  sleepCounter++;
}

void timerInit()
{
  while (RTC.STATUS > 0) {}                /* Wait for all register to be synchronized                    */
  RTC.CLKSEL      = RTC_CLKSEL_INT32K_gc;  /* 32.768kHz Internal Ultra-Low-Power Oscillator (OSCULP32K)   */
  RTC.PITINTCTRL  = RTC_PI_bm;             /* PIT Interrupt: enabled                                      */
  RTC.PITCTRLA    = RTC_PERIOD_CYC32768_gc /* RTC Clock Cycles 32768, resulting in 32.768kHz/32768 = 1Hz  */
                    | RTC_PITEN_bm;        /* Enable PIT counter: enabled                                 */
}

void setup() {
  for (uint8_t i = 0; i < 8; i++)
  {
    *((uint8_t *)&PORTA + 0x10 + i) |= 1 << PORT_PULLUPEN_bp;
    *((uint8_t *)&PORTB + 0x10 + i) |= 1 << PORT_PULLUPEN_bp;
  }
  pinMode(PIN_DEBUG_LED, OUTPUT);
  timerInit();
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
}

void loop() {
  digitalWrite(PIN_DEBUG_LED, HIGH);
  delay(100);
  digitalWrite(PIN_DEBUG_LED, LOW);
  powerOff();
}

void powerOff()
{
  ADC0.CTRLA &= ~ADC_ENABLE_bm; // Disable ADC
  sleepCounter = 0;
  do {
    sleep_cpu();
  } while (sleepCounter < 10);

  ADC0.CTRLA |= ADC_ENABLE_bm;  // Enable ADC
}

```

freemovers, 2024-09-14:
A lot of information about the sleep modes can be found here: 
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/PowerSave.md
Since you are using an ATtiny3216 you have to make sure that you disable input buffer and enable the internal pull-ups for  pins on PORTC as well. If you want to reduce the power consumption even more, then you should replace the delay(100) with another timer (RTC) and have it sleep for that time period.

PM04290, 2024-09-17:
@SpenceKonde i read somewhere that you plan to investigate for Sleep power saving. some remarks to help you in this way :
If hardware have floating input, put them in INPUT_PULLUP have no (signifiant) effect, the only way to stop consumption is to desactivate input
`PORTA.PIN0CTRL = PORT_ISC_INPUT_DISABLE_gc;`
Exemple on ATtiny1616 with all pins floating, in power off sleep mode i have 0.1µA
BUT, it's imposible to go back with friendly function, like pinMode, when wakeup.
if you write a dedicated library for power saving, you may create function to manage pins easly

SpenceKonde, 2024-12-07:
We already have an extension that lets you set this (and the rest of the bits in PINnCTRL)
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Digital.md#pinconfigure 
pinConfigure() note that that document is written to cover both mTC and DxC, which differ in very few ways for the purposes of that discussion

SpenceKonde, 2024-12-08:
issue does not have anything to do with the core, but - much thanks for the bit about the pullup not working like it used to, I'll add that to docs. 

I think my sleeplib is going to be broken up into two or three "tiers", with a basic set of commands to do sleep normally, and then more complicated ones that bring out the true capability of these parts, like sleepWithMillis() (turn off millis right before sleeping after noting down internal values, turns on RTC to keep time in sleep. wakes on normal wakeup source, but otherwise is woken by the RTC (in standby mode, recommended and much more accrate) or the PIT (less accurate, 1 PIT cycle uncertainty)counting each one and then returning to sleep. On wakeup from another source, stops RTC, uses that + number of overflows recorded to calculate how long it has been, adds that to the millis totals, and restarts millis, upon which normal code would resume. That way you can get micros() and non-gimped timekeeping while awake without losing millis record in sleep. And obviously we need the closely related sleepForMillis(uint32_t time); to sleep for that long unless otherwise woken, that sort of thing....

PM04290, 2024-12-08:
i think it's a good way to have a sleep library with friendly using.

SpenceKonde, 2024-12-14:
closing this issue, there is already an open issue to create my sleeplib, and the issues are to-do list

closed 2024-12-14 by SpenceKonde