SpenceKonde/megaTinyCore #1148: Error setting analog reference for Tiny-2

EclipseBoom, 2024-10-04:
Using an Tiny826 chip and the latest core version on Arduino 2.3.3

When setting the analogReference I sometimes get a compilation error "Implicit Definition" or "Undefined symbol" i.e.

`  analogReference(INTERNAL2V048); // set reference to the desired voltage, and set that as the ADC reference.
`

Work Around:

By selecting a different family in the IDE Board Menu such as ATTiny3227... and then reselecting the correct family and chip.  The compile will work properly and will continue to work properly.  Seems like the correct analogReference option set is not being set properly the first time.

SpenceKonde, 2024-12-06:
I think this is a new IDE bug...

SpenceKonde, 2024-12-13:
It just SCREAMS "overly aggressive caching" (hence, stale compiled objects are (trying to) get in_ Things where you had to clear the cache like that were common back in the darker ages, but that had seemed to havebeen put behind us by now. I guess not.
I was very very disappointed with the 2.x IDEs before I was forced to take that time off from this - it was back to the troubled 1.6x days wehere they fix one bug and introduce another, and never manage to release one that doesn't have any bugs  that you consider a showstopper. I still use 1.8.13 - because I know how it behaves, and is free of greviously pathological behavior and has basic functionality intact (other than the security issue), which is more than I can say for any other 1.6.x through 2.x.x that I have tried.... I hope to be pleasantly surtprised when I install the new version. But this isn't fi;l;ing me with optimism.

SpenceKonde, 2024-12-13:
Please do reopen it if you're able to find evidence that it is *not* just arduino's overaggressive cacheing

closed 2024-12-13 by SpenceKonde