SpenceKonde/megaTinyCore #1149: PWM Timer Options must be explicitly set for ATTiny826

EclipseBoom, 2024-10-04:
Using an Tiny826 chip and the latest core version on Arduino 2.3.3

In order to compile the PWM Pins option "No TCD PWM Even on 1-series" must be manually selected.  The core doesn't recognize the Tiny826 as not having TCD by default.  Core tries to compile with references to TCD

Work Around:

Manually select "No TDC PWM" from PWM pin option menu

SpenceKonde, 2024-10-15:
What the heck!? How has this gone unnoticed so long

SpenceKonde, 2024-10-15:
How was this not detected ages ago? This is fucking gross. Everything gated by USE_TCD_PWM should also be enabled only if TCD0 is defined. So how did I miss something? Of course, I need to get the development system working again. Can't compile anything right now.

EclipseBoom, 2024-10-15:
Perhaps this is just in the one (few) CPU definition not the entire core configuration.  I know I've used the 1626 brother part without needing to set these options on previous versions of the core.  Not for a while so a ninja bug may have crept in.

P.S.  Not possessing God-like all-knowing powers IS a tough blow to face.  I've had to deal with it a few times over the years.  I've mostly recovered.  :-)

SpenceKonde, 2024-10-15:
No i 100% a novel bug that should neer have snuck in, i m,erged something I dhouldnt hjave

EclipseBoom, 2024-11-27:
Same situation for ATTiny1626.  Not a surprise but thought I would let you know.

hmeijdam, 2024-12-17:
I just pulled your commits until "Update dirty _tricks.h and when I then compile an empty sketch for the 826, 1626 or 3226 I get this:

```
In file included from E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr\variants\txy6/pins_arduino.h:27:0,
                 from E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr\cores\megatinycore\Arduino.h:713,
                 from E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr\cores\megatinycore\wiring_private.h:32,
                 from E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr\cores\megatinycore\wiring_digital.c:28:
E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr\cores\megatinycore/timers.h:54:15: note: #pragma message: Note: This part does not have a TCD, hence there is no PWM from TCD available.
       #pragma message("Note: This part does not have a TCD, hence there is no PWM from TCD available.")
               ^~~~~~~

```

816 etc no error

EclipseBoom, 2024-12-17:
Seems to be pretty much fixed  In 2.6.10.

I changed my options to the default for 1626 and got the error.  Then compiled again and didn't get it.  

Then tried a bunch of stuff with new sketches.  Different chip groups and even changed to Teensy and back and it continues to work.  So your theory about the GUI somehow caching the incorrect menu options seems to be plausible.

I think this one can be closed.  If it keeps popping up perhaps just a note in the documentation.

SpenceKonde, 2024-12-19:
More PWM fixes going in, keep eyes on this

closed 2024-12-19 by SpenceKonde

EclipseBoom, 2025-03-22:
I know this issue is closed but I just ran into the same "This part doesn't have TCD0" error.  Here are the details:

chip: Attiny1626
core: 2.6.10
ide: 2.3.4
os: Windows 10

Been compiling with the same settings for weeks on this chip.  No issues.  
But today I changed EEPROM retained to EEPROM not retained - and got the error.
Did not actually do the Burn Bootloader operation.
Changed it back and no error.  Changed back to not retained and no error.

Sounds like some kind of initialization error when the setting is changed.