SpenceKonde/megaTinyCore #1158: Wire.usePullups() stopped working after version 2.6.1

jvasileff, 2024-10-29:
As can be confirmed with a multimeter and the following setup code:

```
Wire.usePullups();
Wire.begin();
```

`Wire.usePullups()` stopped pulling `PB0` and `PB1` high with version `2.6.2` or `2.6.3`. This was tested with an `attiny1627`.

MX682X, 2024-10-30:
https://github.com/SpenceKonde/megaTinyCore/blob/f0e59764d741bdff09b670be0d055b5c728b4788/megaavr/libraries/Wire/src/twi_pins.c#L441

could you try the newest version but remove the right part of the #if conditional? I don't have a 2-Series to test

jvasileff, 2024-10-30:
Thanks! Indeed, that does fix `Wire.usePullups()` on 2.6.10. I just did a quick test with a multimeter, I didn't try to actually use i2c with this change.

jvasileff, 2024-10-30:
I've now tested actually using i2c with this change - everything works great.

SpenceKonde, 2024-11-01:
There are CONSIDERABLY more serious issues here. Much more complicated than implied here.

MX682X, 2024-11-01:
How so?
As far as I can see the use Pullups function isn't doing anything in the 2 series.
all we have to do is to #if def the OUTCLR, however I think it would be sensible to just keep it like that - I doubt anyone will end up using I2C pins on a PCB for anything but I2C

SpenceKonde, 2024-12-17:
The fixes I just checked in should correct the issue. Highlighting was fucked on my text editor, so I thought al the stuff  Iremoved was commented out anyway..... 

Also, I am adding to the contributions guide a hard prohibition on if (true == (.....) unless someone can defend that to me.

closed 2024-12-17 by SpenceKonde