SpenceKonde/megaTinyCore #1159: Installation not possible because of certificate

DereIBims, 2024-10-29:
[image]

DereIBims, 2024-10-29:
The certificate expired 7 day ago

hmeijdam, 2024-10-29:
You can try using https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469 as workaround

That worked fine last time when the certificate expired.

per1234, 2024-11-04:
For an update on the situation, see **@SpenceKonde**'s comments on the issue for the same problem as it affects the ATTinyCore platform:

https://github.com/SpenceKonde/ATTinyCore/issues/885#issuecomment-2451770909

SpenceKonde, 2024-12-05:
It is fixed. My main computer failed, so I couldn't get into the website to administer it - I knew I had the certs saved somewhere, but they were in the place they made the most sense to put them. I hadn't thought of that place while looking so I was only finding lots of wrong certs in the place that made almost as much sense to put them. 

And I needed to finish up tax bullshit. And I thought the certs were on my file server along with most of my other data, but it wasn't running and until I got a new pair of scope probes (gave up after failing to find any  within a month) and could see what was wrong with the power supply and track it down to the bad connector and replace it, thus restoring access to the fileserver. Which in fact did not have the cert I needed on it. And I discovered that my file server was failing to negotiate a 1gbit connection to my network switch, and 100 mbits to your bulk data storage is just not acceptable. That I "solved" by getting a 2.5 gbit switch and a 2.5 gbit network adapter, and a new set of cables, so now I get speeds 

So I ordered a M2 SATA enclosure. Received a USB hub that looked just like the enclosure I'd intended to order. Ordered the SSD enclosure, removed SSD from old computer and located the certs. Located certs with search. Then located them manually in the same location on the filesystem on my new computer - in C:/(redacted for security)/Electronics. I'd been sure they were in the parent directory, leading to me spending week+ on one way of getting the certs and a week on another way, neither of which were needed.....

(Also, USB 3.x? My ass is that enclosure USB 3.x. It bottlenecks at USB 2.0 speeds even with all USB 3.x hardware in it's path. Oh well. The SSD is going into the file server for caching after I'm done rescuing stuff from it, after which point the enclosure will be unused until another computer fails. I shouldn't get creeping along at 30 MByte/sec from SSD storage via USB 3.2. Not when I can write to spinning rust over a network cable at >100 MByte/s (2 drive raid 1, aka mirroring, not striping).

I also got my giant folder of toolchain build stuff back. Anyone wanna guess how large that directory was?

twasiluk, 2025-02-16:
Certificate expired Feb 12th 2025. Let me know if I can help.