SpenceKonde/megaTinyCore #1168: Current master branch doesn't compile

jvasileff, 2024-11-16:
The master branch fails to compile due to an incomplete comment in `core_devices.h` added in ff882057402e7:

https://github.com/SpenceKonde/megaTinyCore/blob/f0e59764d741bdff09b670be0d055b5c728b4788/megaavr/cores/megatinycore/core_devices.h#L734

hmeijdam, 2024-11-16:
I get the same error.

As a workaround I reverted these two changes in my Github Desktop copy of the current master branch, and now I can compile again. Maybe give that a try also.

[image]

jvasileff, 2024-11-17:
@hmeijdam thanks, yeah, the fix is to either remove line 734 of `core_devices.h` or revert commit ff8820574

commit b3825605 (2024-12-05): Fix #1168

how did these compile failures not get caught by authomatic testing?
closed 2024-12-05 by SpenceKonde