SpenceKonde/megaTinyCore #1172: When printing continuously , Attiny202 will not print correctly.

dty717, 2024-12-06:
Attiny202 need a delay to print right when printing continuously.

SpenceKonde, 2024-12-06:
Please clarify.

Are you sure it's not clock mismatch? That's often a sign of a clock mismatch - things work for one or two characters in a row, but when sending or receiving continually, it starts giving gibberish for long enough blocks of text. 

Happens when the sender's timebase is running a little fast, so it starts the next start bit when it's counterpart thinks it should still be a stop bit. Do that a few times in a row with no pause, and you inevitably have a framing error and gibberish. 

UART is an ancient standard, dates to the days of dinosaurs and teletypes, however, and this issue with a pileup of characters in succession not being received correctly has long been known, and a solution is now standard on nearly every USART and UART device: You just automatically add a 1-bit-period delay after each byte - this is what the optional second stop bit is for (and recall that the state of a stop bit is a HIGH, also the idle state of the line - so it does, indeed, force the USART to delay each char slightly. 

I wager that clock mismatch is your problem that that the sending device is the one running faster, and thus using 2 stop bits is the appropriate and efficient countermeasure. 
It should also respond to more drastic approaches, which are not recommended like nudging down the baud register until it works better, or lowering the OSCCAL a notch or two in setup to slow down the sending (or equivalently, telling the receiving device to listen at a faster baud rate, assuming it allows specification of arbitrary baud rates) 

They've gotten a lot better since the bad old days of +/-10% with the RC oscillator clock source on old tinyAVRs, these are *much* better, but you still occasionally get a combination that won't play well together. In the section of the core docs covering the extensions to the Serial class, I go into this in more detail... (and sometimes, a slow serial adapter. Notice that whereas old serial adapters always had a visible crystal dedicated to the serial adapter, giving them very accurate serial timing, most serial adapters today just use an internal RC oscillator (for example, the original CH340G needed a crystal, and spec'ed timing accurate to within 0.5%. The now-more-abundant, cheaper ones only spec +/-2% accuracy for their own baud rate calculations). same applies if you have two avrs talking (especially if one of them is a classic, see the serial doc; due to technical limitations of the time, classic AVR serial ports have another kind of error, which is negligible on modern AVRs

As this is inherent to the protocol and the solution known and documented, in the absence of evidence otherwise, I see no bug here. Please reopen if such evidence is discovered.

closed 2024-12-06 by SpenceKonde

dty717, 2024-12-06:
This is my code, not print exact in Attiny202 with default setting.
```

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
}
int inChar;
void loop() {
  Serial.print("get char:");
  Serial.print(inChar);
  Serial.print("\r\n");
  Serial.println("\r\nHello world!");
  Serial.println("Hello world!");
  delay(1000);
}
```

SpenceKonde, 2024-12-14:
Have you tried what I suggested? I gave you one good suggestion:

void setup() {
  // Open serial communications and wait for port to open:
  //RIGHT WAY - Use 2 stop bits to ensire that the receiver can still receive it even when sending continually with a slgihtly faster timebase. 
  Serial.begin(115200,SERIAL_8N2);
  //while (!Serial) {
  //  ; // wait for serial port to connect. Needed for native USB port only
  //} These don't have native USB so why have that up there?
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
}
int inChar; /* Poor style, put globals at the top of the file */ 

void loop() {
  Serial.print("get char:");
  Serial.print(inChar); //This will always print 0. 
  Serial.print("\r\n");
  Serial.println("\r\nHello world!");
  Serial.println("Hello world!");
  /*delay(1000);*/
  /* With that delay commented out, it will be continually sending at the full baud rate. 
   *  If the receiver's  timebase is even a little slower than the sender's, it cannot possibly work indificitely. 
   *  If the sender is sending characters 0.2% fast, but is sending continually without ever pausing, eventually that 
   *  0.2% adds up to enough that it's seeing a start bit where the stop bit should be, then interpreting the next low 
   *  as the start bit of the next frame and now you receive garbage.
   *  In practice, there are tricks that the receiver side uses to give itself a ltitle for clock skew in that direction
   *  (I think it gets like a third of a bit periodf per frame because it is reset and waiting for the next start bit 
   *  as soon as it's done sampling the stop bit, in the middle of  that bit's bit period 
   *  
   *  The standard solutionm, used around the world since before I was born is to use 2 stop bits :-P 
   */
}

And two bad ones! 
This is one bad solution (blind-tune the osc down one notch, since we know the tiny202 sending the data is too fast for the receiver ( assuming my diagnosis is correct, which I'm very confident in))
```

void setup() {
  // Open serial communications and wait for port to open:
  // BAD WAY 
  _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA,(*(uint8_t*)((uint16_t)(0x1100 + 24))-1)); //0x1100 + 26 for 20 MHz system clock
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only 
  }
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
}
int inChar;
void loop() {
  Serial.print("get char:");
  Serial.print(inChar);
  Serial.print("\r\n");
  Serial.println("\r\nHello world!");
  Serial.println("Hello world!");
  delay(1000);
}
```

This is also a bad solution:
```
void setup() {
  // Open serial communications and wait for port to open:
  // BAD WAY #2        
  Serial.begin(113600); 
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
  Serial.println("\r\nHello world!");
}
int inChar;
void loop() {
  Serial.print("get char:");
  Serial.print(inChar);
  Serial.print("\r\n");
  Serial.println("\r\nHello world!");
  Serial.println("Hello world!");
  delay(1000);
}
```

I know the good one will work, and is the only one I recommend, but you can use the other two to proove to yourself what's going on). I suspect the other two will also work, though they might need tuning to the specific specimens you're working with.

SpenceKonde, 2024-12-18:
Please post a screenshot of all the tools menu options you have selected and I'll try to dig out some 202s. Matybe even rthrow a couple onto the digikey order...

SpenceKonde, 2024-12-18:
OH!!! If you have a scope trace of the TX line of the 202 that's misbehaving, that'd probably blow this open...

dty717, 2024-12-18:
The download tool is raspberry pi pico with [debugprobe uf2](https://github.com/raspberrypi/debugprobe) as usb to serial,
[image]

Code is complied by arduino version 1.8.13 and downloaded by [pymcuprog](https://github.com/microchip-pic-avr-tools/pymcuprog).
[image]

Realterm is used to receive data.
[image]