SpenceKonde/megaTinyCore #1173: Error when trying to use the RESET pin as GPIO (no longer an option?)

replaysMike, 2024-12-07:
Is there a specific reason there is no longer an option for reassigning the reset pin to be used as GPIO?

In my case I'm using an ATTINY3224, and I need to use the reset pin (PA0) as GPIO input but not sure how I'd go about doing that (newbie).

```cpp
int val = analogRead(PIN_PA0);
```

Assertion:
```
In function 'check_valid_analog_pin',
    inlined from 'analogRead.constprop' at C:\Users\mikeb\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\cores\megatinycore\wiring_analog.c:321:5,
    inlined from 'ReadButtonInput' at C:\Users\mikeb\OneDrive\Documents\Arduino\Binner\CabinetAssembly_ADA\CabinetAssembly_AdafruitGFX\CabinetAssembly_AdafruitGFX.ino:83:31,
    inlined from 'loop' at C:\Users\mikeb\OneDrive\Documents\Arduino\Binner\CabinetAssembly_ADA\CabinetAssembly_AdafruitGFX\CabinetAssembly_AdafruitGFX.ino:55:35,
    inlined from 'main' at C:\Users\mikeb\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\cores\megatinycore\main.cpp:56:9:
C:\Users\mikeb\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\cores\megatinycore\wiring_analog.c:58:11: error: call to 'badArg' declared with attribute error: 
           badArg("analogRead called with constant pin that is not a valid analog pin");
```

hmeijdam, 2024-12-07:
That is to protect newbies from accidentally disabling the UPDI (PA0) pin, so they won't be able to program the part again. 
You need a UPDI-HV (high volatge) programmer to revert that pin back to UPDI

You can enable "expert mode" in your boards.txt file from MegatinyCore

You will find thiese menu items commented out

```
#----------------------------------------#
# UPDI/RESET PIN CONFIGURATION           #
#________________________________________#
#Expert mode - if you understand what these mean, uncomment them to enable.
#there are a few lines like this in platform.txt too that can be uncommented to enable for uploads, not just when burning bootloader
#atxy4.menu.resetpin.UPDI=UPDI (no reset pin)
#atxy4.menu.resetpin.UPDI.bootloader.resetpinbits=01
#atxy4.menu.resetpin.reset=Reset (DANGER - Bricks chip w/out HV UPDI programmer!)
#atxy4.menu.resetpin.reset.bootloader.resetpinbits=10
#atxy4.menu.resetpin.gpio=GPIO (DANGER - Bricks chip w/out HV UPDI programmer - Use PCHV)
#atxy4.menu.resetpin.gpio.bootloader.resetpinbits=00
```
Uncomment them to enable the menu options to disable UPDI. At your own risk.

SpenceKonde, 2024-12-07:
Yup. There's the snazzy new alt reset pin option on the 2-series - you can move reset to PB4. Unfortunately, since the 14-pin parts don't have a PB4, it doesn't help you there.

I think the HV trick with the 12v doorbell battery `*` and a series resistor, where you manually apply the 12v to reset still works on the tiny2 - I don't see anything on DS40002345A-page 446 (rev A of the 3224~7 datasheet, the latest one I'm aware of, study that page) mentioning the change that happened to break that on Dx and Ex parts that let you override UPDI, so even without sending a KEY, I think the 12v pulse will set the pin back to UPDI mode until a POR. They warn about that in the middle of the page I referenced.  Obviously that pissed off customers (cause like what do you do? you power chip straight from battery so you have nothing in the middle drawing added quiescent current, chip in sleep power down, and you put it in a sealed plastic case like every other manufacturer. So when static electricity triggers the updi mode instead of the GPIO mode (which may be essential for product functionality), unless the loss of GPIO also caused it to fail to go into sleep mode, such that it could drain it's battery in time for you to notice, assuming you recharged it once more after that - and discovered it working again. If sleep still worked, but without PA0 in GPIO, basic functionality didn't, unless someone online told you to open it and disconnect and reconnect the battery, you'd be throwing it out or sending it back for a refund or warranty repairs). So on Dx/Ex, so they put in the 50ms limit between the HV pulse and a key before it would give up and go back to the mode set by the fuses. 
This section is about the problem of the pin being configured in a way soas to prevent the pin from being raised to 12v is present: 
> When the RESET Pin Configuration (RSTPINCFG) bit in FUSE.SYSCFG0 is ‘0x0’, the RESET pin is configured as
> GPIO. To avoid a potential conflict between the GPIO actively driving the output and a UPDI high-voltage (HV) enable
> sequence initiation, the GPIO output driver is disabled for a minimum of 8.8 ms after a System Reset.
> It is always recommended to issue a Power-On Reset (POR) before entering the HV programming sequence

That's not the one that breaks the ghetto HV prog method
Note that I don't know why that's such an issue! The output drivers on PA0 used as GPIO on modern tinyAVR are pathetically weak, and they don't specify a abs max current that's any different for reset. They do spec output driver strength though, which it turns out is terrible. it looks to me like even with the pin driving low, you could push the pin all the way up to 12v and stay within the abs max current per pin >.> But they seem concerned about it, and they know better than me. (dear lord I hope they do. Sometimes.... )

But yeah, you do know that whatever pin is the pin gets the HV programming pulse (PA0 on modern tinyAVR, PF6 on Dx/Ex, and on classic AVRs it was, uh, I dunno, whichever pin was closest to where the "RST" dart landed when they designed that family - not that knowing the port number helped you find it back then, as the pins weren't arranged in a consistent order. You can see the legacy of that on our tinyAVR layout where PORTB is inexplicably backwards) - the HV tolerant pin, if it can be configured for other functions (on tinyAVR, UPDI or GPIO), any of those functions that involve the pin acting as an output (including the chip's replies in UPDI - while playing with hardware for programming, you could actually tell which end made which pulse, because the tinyAVR's drivers were so weak that their LOWs didn't go as close to zero, and with an inappropriate resistor, it is possible to make UPDI fail by way of the chip failing to pull the line low enough to make plain just what a miserable output it is. At 5v, normal I/O pin specs 15mA with pin voltage no lower than 4v. On the reset pin, it only guarantees  >=4V if the current drawn from it is under 0.5mA. (and it's equally weak pulling to ground) Yup, 2k ohm output impedance when the UPDI/reset pin is set GPIO. The output impedance of a normal pin (calculated from same data) is more like 75 ohms. When it sank in just what an awful GPIO pin you got (around when I noticed they'd axed it from the DA) from turning classic or tinyAVR reset pins into GPIO, I kinda wondered why they'd even bothered with the option to use it as an output, considering what a poor job of being an output it does.

Like, if it's UPDI, you can read the pin as an input without issue as long as whatever you're reading doesn't look like UPDI commands. It's only making it an output that requires fusing it to GPIO, and it's piss poor as an output...

On Dx/Ex (DD and later let you set UPDI pin as GPIO, and Reset as GPI), and need HV pulse on Reset (which isn't the same pin on those) you need to send the KEY to enter NVMPROG mode within 50 ms of the HV pulse or something like that (Also, the reset pin cannot be a GPIO, only an input as noted above - but the UPDI pin is a normal strength pin, not gimped like an HV-tolerant pin.

`*`  doorbell remotes commonly use these 12v batteries made from 8 stacked up watch batteries. Remote control doorbells, regardless of how they're priced, use the cheapest, crappiest receiver topology, and are largely the reason these devices are so utterly abysmal.

SpenceKonde, 2024-12-07:
Not a defect in core, hence closing as issue

closed 2024-12-07 by SpenceKonde

replaysMike, 2024-12-07:
@hmeijdam thank you! That was the info I needed and couldn't find. I do have an HV UPDI programmer so I'm not worried about that.

replaysMike, 2024-12-10:
@SpenceKonde re-read your message as it seems like it might be relevant to my current issue.

> Like, if it's UPDI, you can read the pin as an input without issue as long as whatever you're reading doesn't look like UPDI commands. It's only making it an output that requires fusing it to GPIO, and it's piss poor as an output...

As an input, how do I get past the compile error with megatinycore - "analogRead called with constant pin that is not a valid analog pin"? I don't see any obvious defines/assertions that can be bypassed.