SpenceKonde/megaTinyCore #1187: PWM frequency change for pins PA3 and PA5

ltwin8, 2024-12-30:
hello, i would like to use the Pins PA3 and PA5 for outputting PWM. 

I would like to run at 1MHz to 8MHz internal clock (the lower the better, power saving) and i would like to output a squarewave on pin PA3 with roughly 50% dutycycle at a frequency of about 100kHz at PA3, i did play around with the TCB1.CTRLA setting it to 00 for the prescaler but it did not help, i looked up on datasheet p16 that its source is tcb1.

can i somehow set the resolution of the pwm to 1 bit? that way i could just toggle the pin right? can someone provide me with an example snippet?

additionally i would like to output a pwm of 8 bit resolution on PA5 at somewhere above 5kHz and below 30kHz, please if possible give me a code snippet here too, i will try to figure the rest out myself when i can work of a functioning code

ltwin8, 2025-02-08:
any chance you can help me @SpenceKonde 

is there any alternative to AnalogWriteFrequency like in your megacoreX?

SpenceKonde, 2025-05-09:
I think you want this document? https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/libraries/megaTinyCore/examples/TCA0Demo2/TCA0Demo2.ino

SpenceKonde, 2025-05-24:
And to expand onthat:

A function to adjust the PWM frequency is not planned, because we have such rough control over the PWM frequency as it is. At best, you can get a frequency within a factor of 2 of the target. Sometimes that's not even possible (TCA at highish prescalers, where they're only every power of 4). 

Of course, you can imagine one might write a more complex analogWrite() where you had the timer running at a much lower prescaler to a higher value, and were scaling up the values passed in software) - that's a job for a third party library, however, because it masks fundamentally different tradeoffs than analogWrite() (most notably, I suspect the overhead would be markedly higher - and it might be hard to do without division) 

analogWrite() is meant to do what it always was - give you PWM at a frequency fast enough that you won't get flicker, and slow enough that you can drive small MOSFETs without added gate driver ICs ( We do 480 to 1500 Hz across all the supported speeds and cores, with the more extreme values being because that's what's forced on us by the hardware if we don't want to have to do fucked up math. 

In all cases, though the you may change the prescaler for a PWM timer without breaking the PWM, other than as an indirect consequence of the change in frequency, without having to take full control with takeOverTCA() and (unless it's the millis timer, there's menu options to choose a timer you aren't using.)

On all tinyAVR-series parts, NO TCB IS EVER USED TO GENERATE PWM. They are too limited in number and too useful to waste on one shitty pwm channel. We generate 6 8-bit PWM channels by using the TCA in split mode. 

There is a menu option to select which pins to use.

That menu option is broken in release, but fixed in github version.

closed 2025-05-24 by SpenceKonde