SpenceKonde/megaTinyCore #1188: Why is UPDI pin having Vcc as output?

tobivan, 2024-12-30:
Hi there,

simple circuit: LED at PA1 :-)
Trying some blinky-stuff, and I wonder why the UPDI pin is showing +Vcc 
To save some energy, I took this function and I thought that PIN0 was the UPDI pin:
```
void disableAllPeripherals(void) {
  PORTA.DIRCLR = PIN0_bm; //  set port A0 to input
  //PORTA.DIRCLR = PIN1_bm; //  LED
  PORTA.DIRCLR = PIN2_bm; //  set port A2 to input
  PORTA.DIRCLR = PIN3_bm; //  set port A3 to input
  PORTA.DIRCLR = PIN6_bm; //  set port A4 to input
  PORTA.DIRCLR = PIN7_bm; //  set port A5 to input        
        
  PORTA.PIN0CTRL = PORT_ISC_INPUT_DISABLE_gc; // disable input buffers
  //PORTA.PIN1CTRL = PORT_ISC_INPUT_DISABLE_gc;
  PORTA.PIN2CTRL = PORT_ISC_INPUT_DISABLE_gc;
  PORTA.PIN3CTRL = PORT_ISC_INPUT_DISABLE_gc;
  PORTA.PIN6CTRL = PORT_ISC_INPUT_DISABLE_gc;
  PORTA.PIN7CTRL = PORT_ISC_INPUT_DISABLE_gc;
}
```

What am I missing? Do I need to connect UPDI pin if not needed?
Thanks a lot!

hmeijdam, 2024-12-30:
You need to first set the UPDI pin as GPIO pin to be able to use it as GPIO pin. 
After doing so, you will need to have a HV programmer to ever set it back to UPDI and reprogram the chip. Which is the reason why this choice is hidden/commented-out as "expert-mode" in the boards.txt file.

tobivan, 2024-12-30:
Thanks for your answer.
I don't want to use it as a GPIO pin. But why can I read VCC voltage on the UPDI pin? If it is a "programming pin" I would have expected it to be either input or connected to GND, but in no case to Vcc.

hmeijdam, 2024-12-30:
Because the UPDI is HIGH in it's idle state and it has a internal pullup resistor. That's what you are measuring. If you connect a led to it, you will see it lights up a little (current flowing through the pullup resistor)

hmeijdam, 2024-12-30:
> But unless not connected, it cannot draw any current, can it?

"Unless not connected" means it is connected. 

**_If_** not connected it cannot draw current.

closed 2025-05-24 by SpenceKonde