SpenceKonde/megaTinyCore #1191: ATtiny1626 Serial programming issues

WhoTookMyNicks, 2025-01-04:
Hello,

I tried to upload the Blink sketch to the ATtiny1626 and got errors.

Firstly, I have burned the bootloader by setting the Programmer to SerialUPDI 230400. Everything goes fine.
Next, while uploading a sketch I get this error:

> Arduino: 1.8.19 (Windows 10), Board: "ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406, ATtiny1626, 20 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), PB0-2, PA3-5, 1-series: PC0/1 (default), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""
> 
> Sketch uses 674 bytes (4%) of program storage space. Maximum is 16384 bytes.
> 
> Global variables use 4 bytes (0%) of dynamic memory, leaving 2044 bytes for local variables. Maximum is 2048 bytes.
> 
> SerialUPDI
> 
> UPDI programming for Arduino using a serial adapter
> 
> Based on pymcuprog, with significant modifications
> 
> By Quentin Bolsee and Spence Konde
> 
> Version 1.2.3 - Jan 2022
> 
> Using serial port COM19 at 230400 baud.
> 
> Target: attiny1626
> 
> Set fuses: ['0:0b00000000', '2:0x02', '6:0x04', '7:0x00', '8:0x00']
> 
> Action: write
> 
> File: C:\Users\usr\AppData\Local\Temp\arduino_build_808814/Blink.ino.hex
> 
> Traceback (most recent call last):
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 286, in <module>
> 
>     main()
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 128, in main
> 
>     return_code = pymcuprog_basic(args, fuses_dict)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/tools/prog.py", line 201, in pymcuprog_basic
> 
>     args_start)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\pymcuprog_main.py", line 545, in _start_session
> 
>     backend.start_session(sessionconfig)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\backend.py", line 362, in start_session
> 
>     sessionconfig.interface_speed)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\programmer.py", line 83, in setup_device
> 
>     options=self.options)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\nvm.py", line 42, in get_nvm_access_provider
> 
>     accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\nvmserialupdi.py", line 54, in __init__
> 
>     self.avr = UpdiApplication(port, baud, self.dut)
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\serialupdi\application.py", line 79, in __init__
> 
> An error occurred while uploading the sketch
> 
>     datalink.init_datalink()
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\tools\libs\pymcuprog\serialupdi\link.py", line 44, in init_datalink
> 
>     raise PymcuprogError("UPDI initialisation failed")
> 
> pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed

What I am doing wrong?

hmeijdam, 2025-01-04:
> What I am doing wrong?
Not telling how you wired everything up when you burned the bootloader compared to how you wired it up after you installed the bootloader. And what options you selected in both cases in the tools menu.

WhoTookMyNicks, 2025-01-04:
Yup, my question is a little bit empty.
I used this schematic:
https://oshwlab.com/wagiminator/y-attiny3216-development-board-smd_copy_copy

When switched to UPDI I burned bootloader and get this:

> SerialUPDI
> UPDI programming for Arduino using a serial adapter
> Based on pymcuprog, with significant modifications
> By Quentin Bolsee and Spence Konde
> Version 1.2.3 - Jan 2022
> Using serial port COM19 at 230400 baud.
> Target: attiny1626
> Set fuses: ['0:0b00000000', '1:0x00', '2:0x02', '5:0b11110111', '6:0x04', '7:0x00', '8:0x00']
> Action: erase
> Pinging device...
> Ping response: 1E9429
> Setting fuse 0x0=0x0
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x1=0x0
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x2=0x2
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x5=0xf7
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x6=0x4
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x7=0x0
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Setting fuse 0x8=0x0
> Writing literal values...
> Verifying literal values...
> Action took 0.03s
> Finished writing fuses.
> Chip/Bulk erase,
> Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
> Memory type flash is always erased
> Memory type lockbits is always erased
> ...
> Erased.
> Action took 0.01s

When I switched to UART I got the error from above. In a result, you have all the settings. It is in the first few lines.

hmeijdam, 2025-01-04:
That has a serial to UPDI chip already on the board, so you don't need the bootloader.

WhoTookMyNicks, 2025-01-04:
According to:
https://oshwlab.com/wagiminator/y-attiny3216-development-board-smd_copy_copy
It does.

Nevertheless, I can not upload a sketch.

hmeijdam, 2025-01-04:
This should work 
[image]

And of-course you set the switch on the board to UPDI

WhoTookMyNicks, 2025-01-04:
I missed this with a switch. It was on UART all the time during the upload of the sketch. Everything works for now.

Thank you.

hmeijdam, 2025-01-04:
Ok. You should close the issue.

closed 2025-01-10 by SpenceKonde