SpenceKonde/megaTinyCore #1196: attiny3226, calling millis to time events causes TCA0 to loose counts.

ieb, 2025-01-10:
I think its my code, but I am after some guidance.

If I use code like this if find that Timer A0 compared to an external time pulse looses counts. At least, as best I can tell they loose counts.

            void loop() {
              static unsigned long lastCall=0;
              unsigned long now = millis();
              if ( now-lastCall > 1000 ) {
                  reportOnFrequencyMeasurement();
              }
            }

However if I do 

            void loop() {
                reportOnFrequencyMeasurement();
                delay(1000);
            }

The pulse width measurement after tuning the internal clock oscillator is within 0.2% of the pulse width measured against a HC-49 crystal. Also verified on a scope. I don't see the same behaviour running the code on a 328p using millis in this way.

I've tried a few techniques to measure the pulse width, all using ISRs including:
1. TCB0.CTRLB |= TCB_CNTMODE_FRQ_gc, route PIN_PC2 using events to the timer capture event and then to an ISR.
2. run take over TCA0, count freely at 250KHz and capture the counter after 50 edges keeping an eye on overflows of the TCA0.CNT value.
3. Dumb way, count 50 edges and capture the change in micros() every 50th edge, this works well enough on a 328p with the first loop.

PER_CPU appears to be running at 16MHz, I tried DIV/2 on TCB0 but that didn't make a difference.
Input pulse frequencies in the range 200Hz to 4KHz.

millis is on TCB1

Question: Whats the right way of implementing a non blocking event loop in code on a attiny series 2 so that the timers don't loose counts ? 

I've read all the extras (excellent docs) but havent spotted anything to indicate calling millis excessively would delay counters.

The Attiny3226 is from a genuine batch bought from DigiKey in 2021.

hmeijdam, 2025-01-10:
As soon as you configure a timer to start running, it will continue on it's own, regardless what your code (or the millis() code) does with another timer. 

Most likely the problem is your sketch, that you did not share.

SpenceKonde, 2025-01-10:
Please post code in it's entirety which reproduces this problem. 

If millis is losing counts within a situation where you are not making this eventuality unavoidable as described, then this is a highest-of-the-high priorities.  (essentially: In any situation in which either the runtime of an ISR, or a period during which interrupts are disabled which exceeds 1 ms, millis will have a chance equal to ((length of time interrupts blocked for) - 1 ms)/1ms = L of losing 1 ms (doing math in the ordinary, not integer or floating point, realms), if this L > 1 (t > 2ms) the integer portion is the number of milliseconds always lost, and the decimal portion the chance of losing an additional ms. 

Keeping interrupts disabled for excessive lengths of time without consequence is not a design goal of these cores; an interrupt that exceeds a runtime of 1ms is pathological. That which disables interrupts for longer 1ms may be pathological or it may be required; in these circumstances it is the responsibility of the code to do the necessary corrections. I know I've atleast thought about implementing time correction in the one case I know of that is NOT pathological and still; requires disabling interrupts for longer - tinyNeoPixel (the same is true of anything that monopolizes the CPU to bitbang a transfer exceeding 1ms in length, but few of these take that long - othert than neopixel driving;

ieb, 2025-01-11:
Thank you for the pointers.
The code below produces the first set of output with "blocking" and "non blocking".  This code does *not* reproduce the problem. My working copy is at [1]. 

When this code is integrated with a larger project[2]  the problem appears. I am still trying to reproduce the behavior in the larger project but please don't waste any time looking at it until I can reproduce with something simple. Quite unlikely I've uncovered a problem in megaTinyCore.

Based on the pointers you have given, I suspect one of the subsystems is disabling interrupts for too long, so I am bisecting to identify which one. I use SPI (mcp2551 CAN) + OneWire (your branch) as well as ADCs, but not a lot else low down. 

The timerB0 code using the frequency measurement mode and events can be seen at [3], however I found this to not produce consistent results with a clean accurate input signal. Not sure why yet. 

1 https://github.com/ieb/N2KEngine/blob/wipAttiny3226/testFrequencyCapture/src/main2.cpp
2 https://github.com/ieb/N2KEngine/tree/wipAttiny3226
3 https://github.com/ieb/N2KEngine/blob/wipAttiny3226/testFrequencyCapture/src/main.cpp
            
            #include <Arduino.h>  
            
            
            /*
             * Targeting a Attiny3226 using TCA0. Would work on most Attiny chips.
             *  This method uses a free running timer at 250KHz divided down from the peripheral clock.
             *  Each edge is counted by an ISR. When the count reaches 50 the value of the counter is captured
             * representing a count of the number of 250KHz cycles over 50 input pulses.
             *
             *  Provided the uint16_t counter does not overflow, which it will only do less than about 200Hz.
             *  The methods appears consistent, stable measurements for stable frequencies tracking changes in
             *  50 cycles.  Accuracy after tuning the internal oscillator measured at +-0.2%., ie 2 RPM at 1K RPM.
             * 
             *  
             *
             */
            
            
            #if defined(MILLIS_USE_TIMERA0) || defined(__AVR_ATtinyxy2__)
              #error "This sketch takes over TCA0, don't use for millis here.  Pin mappings on 8-pin parts are different"
            #endif
            
            uint8_t InputPin = PIN_PC2;
            
            void setup() {
              Serial.begin(19200);
            
              // setup PIN_PC2 to generate an interrupt
              //pinMode(PIN_PC0, INPUT_PULLUP);
            
              PORTC.PIN2CTRL |= PORT_ISC_FALLING_gc | PORT_PULLUPEN_bm ; // pullup on the input pin
            
              // Counter needs to be setup with clock/64 and overflow interrupts running
              takeOverTCA0();           
              
              // 16000000/64 = 250KHz 250000/2^16 =  3.8146972656 overflows/s
              TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL0_bm | TCA_SINGLE_CLKSEL2_bm; // div 64
              TCA0.SINGLE.CTRLB = 0x00; // normal, just want the counter, no output.
              TCA0.SINGLE.CTRLC = 0x00; // nothing to set
              //TCA0.SINGLE.EVCTRL = 0x00; // disable event actions
              TCA0.SINGLE.PER = 0xFFFF; // Count all the way up to 0xFFFF
              TCA0.SINGLE.INTCTRL = TCA_SINGLE_OVF_bm; //enable overflow interrupt
              TCA0.SINGLE.CTRLA |= TCA_SINGLE_ENABLE_bm; //enable the timer keeping the prescaler.
            
            
              Serial.println("Started");
            }
            
            
            volatile uint8_t edges = 0;
            volatile uint8_t slot = 0;
            volatile uint8_t capturedSlot = 0;
            volatile uint16_t capturedCounts[2] = { 0,0 };
            volatile uint16_t capturedOverflows[2] = { 0,0 };
            volatile uint16_t timerISRCalls = 0;
            volatile uint16_t portISRCalls = 0;
            
[... 74 lines cut ...]
            }
            
            
            // Interrupt on edge and capture the counter ever 50 edges.
            ISR(PORTC_PORT_vect) {
              uint8_t flags = PORTC.INTFLAGS;
              if ((flags & 0x04) == 0x04) {
                portISRCalls++;
                edges++;
                if ( edges == 50 ) {
                  capturedCounts[slot] = TCA0.SINGLE.CNT;
                  capturedOverflows[slot] = timerISRCalls;
                  capturedSlot = slot;
                  slot = (slot+1)&0x01;
                  edges = 0;
                }
              }
              PORTC.INTFLAGS = flags;
            }
            
            // interupt on overflow.
            ISR(TCA0_OVF_vect) { 
              timerISRCalls++;
              TCA0.SINGLE.INTFLAGS  = TCA_SINGLE_OVF_bm; // Always remember to clear the interrupt flags, otherwise the interrupt will fire continually!
            }

ieb, 2025-01-11:
Found the problem. 
I was writing to EEPROM too frequently due to a bug introduced by refactoring which dropped a check to see if what was being written had changed.  Probably going into https://github.com/SpenceKonde/megaTinyCore/blob/4d0d75660ccfa72de79c9c4f15a8cd17c9f0ed16/megaavr/libraries/EEPROM/src/EEPROM.h#L74 often enough to cause problems elsewhere (hope the chip isnt toast now, must have been written to millions of times).

Fixed my code and learnt something. Thanks for the guidance.

closed 2025-01-11 by ieb