SpenceKonde/megaTinyCore #1209: board manager installation impossible because drazzy.com certificate expired

fritzlb, 2025-02-22:
The TLS certificate on drazzy.com expired 12th February 2025 making an installation using the board manager impossible.

lupusworax, 2025-02-23:
sadly not working for me, cant find ATTinycore after adding the source to board manager.

fritzlb, 2025-02-23:
I did test the latest IDE v2 from flatpak (Linux) as well as 2.3.4 on Windows 11, both seem to be working just fine

in any case this is only a workaround, the TLS certificate on drazzy.com definitely needs to be renewed

lupusworax, 2025-02-23:
> I did test the latest IDE v2 from flatpak (Linux) as well as 2.3.4 on Windows 11, both seem to be working just fine
> 
> in any case this is only a workaround, the TLS certificate on drazzy.com definitely needs to be renewed

ok reason was that  I already got the attiny core from the old link, on another pc it worked right away, thanks a lot for the workaround!

pa3axa, 2025-02-25:
Hi,

what might work is change the briefly date on your computer before the certificate expiry date of 12-02-2025.
download and install using the board manager. But please fix the certificate.