SpenceKonde/megaTinyCore #1213: Cannot set lock bit on ATTiny814

eRadiofan, 2025-03-11:
This might be down to my lack of knowledge but I am having difficulty setting the lock bit on this 814:

`.../2.6.10/tools/prog.py -t uart -u /dev/ttyUSB0 -b 230400 -wd 1 -d attiny814 --fuses 10:0x00 -v`

Gives the following error:

`ValueError: 1 bytes of data at offset 10 is outside the boundaries of 'fuses' with size 10`

Can someone tell me what to do?

hmeijdam, 2025-03-11:
I am using a different command to lock and unlock. This is what I use.
to lock the chip:
`avrdude.exe -c serialupdi -p t204 -P COM10 -U lock:w:0x00:m `

to unlock the chip:
`avrdude.exe -c serialupdi -p t204 -P COM10 -e -U lock:w:0xC5:m `

I am using a Attiny204 in combination with a serial adapter changed into a UPDI programmer.

eRadiofan, 2025-03-12:
Thanks, I am familiar with avrdude but I was not aware that it works with serialupdi.

In the meanwhile, I found another way:

`pymcuprog -t uart -u /dev/ttyUSB0 -d attiny814 write -m lockbits -l 0x00`

Since this is the official Microchip utility, I shall be using this method from now on.

SpenceKonde, 2025-05-07:
I think theres a way to get serialUPDI to do that, It's based off old pymcuprog code. I don't recall offhand how...

hmeijdam, 2025-05-08:
You mean something else than what I mentioned in my [response](https://github.com/SpenceKonde/megaTinyCore/issues/1213#issuecomment-2714365732)?

SpenceKonde, 2025-05-24:
Yes, that's what I said, with SerialUPDI. your response shows avrdude, I know that it can be added easily if it's not - to the extent that it's not there, it's because no frontend was made. The rest is there, and mechanically, its the same as fuses. 

Maybe we decided that wasn't a useful feature, because you don't normally set them during development.... and someone deploying with the Arduino IDE on commercial scale is a nutcase, and you don't set lockbits if you're not commercial. So the use case isn't really clear to me, and the question almost universally comes from people who are worrying about code security at the start of the process, rather than later, when they have code to secure.

closed 2025-05-24 by SpenceKonde

eRadiofan, 2025-05-24:
I got the help I needed, thanks.

However, I am a bit disappointed that the version of prog.py included with megaTinyCore has been modified deliberately to prevent the setting of lockbits.

There is perhaps a group of users you are not considering. People like myself, sell maybe a few tens of items to a very niche market. I build and devices one at a time. We do like using the Arduino IDE and we do care about IP, so we lock the small number of devices we sell. FYI for the next time you make such decisions.

jvasileff, 2025-05-24:
> What is the point of setting lockbits if the GPL license requires you to provide your sourcecode together with the executable if you use anything of the GPL licensed MegaTinyCore?

Hah, good point! Although there may be a need to protect secret keys in some cases.

SpenceKonde, 2025-05-25:
It's LGPL 2.0 not GPL 3.0. I thought some of the provisions in LGPL 3.0 were potentially impractically demanding, and as far as I can tell, the core should be using an LGPL not a GPL. Regardless, I don't think anyone has been sued for not sharing their source code when using Arduino to build products (though I've seen several projects where their negligent use of an Arduino for a life-safety-critical application (ex, car nuts) lacking appropriate safeguards

I will remind everyone I said I didn't remember if it was possible. I still don't know off hand. If it doesn't work, then we never implemented the means of passing that command from prog.py to the internals. I know that it's a feature of dubious usefulness, because even most people who develop on Arduino use third party upload tools (some actually written from scratch by third party vendors) for production, due to the desire to put serial numbers sequentially without accidentally double-assigning any. And maybe you use the same chips and boards in two configurations, one of which omits the most expensive part mounted on the board or replaces it with a less expensive pin compatible one - or there's some feature that you can charge some customers an extra 50% for, even though it doesn't need any extra hardware. Those programs would be expected to have facility to upload specific content to the BOOTROW, for use with PDID (my plan for implementing BOOTROW support is to sort of poke fun at the whole thing with a library that lets you read and load it from application (hence requiring eitther bootloader or permit flash write to anywhere enabled; the latter makes the first half k of flash a faux-bootloader and stuffs a few instructions in there which can be jumped to from the app, and hence execute as bootloader (though, in production, you'd likely not use something like that  use PDID if you could and your bootloader would set BOOTRP before running user code) 

In any event, it's no time for sadness by those wishing to withhold their code from their customers, Starting on the AVR DU and EB-series, there's now a feature called "PDID", "Programming and Debug Interface Disable", activated by using UPDI to write a two-byte-wide fuse to a magic number. Once this is done, following the end of the UPDI session, UPDI can never be used on that device again. A bootloader, if you put one on, can do whatever it's able to do. Not only did the DU and EB get it (and obviously the safety first, second, third, and fourth "SD-series" got it), while the DB folks were doing the major die rev to fix almost all of the numerous errata on those parts, the DA team was instead rev'ing he DA die to add on the PDID option (that's what the S suffix on AVR DA's means), and left all the silicon bugs in place. So PDID seems to be important to them, and it lets you do something never before possible on an AVR - use a feature as designed which made it impossible to alter the code, The BOOTROW is also growing over time. 

The SD-series is a new AVR series for extremely life safety critical applications. You know, think... dual core. The two cores execute the same instructions, and the results are compared, and the error controller signalled if there's a mismatch. The ram is ECC. No plans to support on DxCore, though if people started asking, they'revery similar to a Dx. But