SpenceKonde/megaTinyCore #1227: Ability to disable printf

Avamander, 2025-05-03:
I'm trying to fit a sketch on an ATTiny817 Xplained Mini board and it has been quite annoying so far.

SpenceKonde, 2025-05-07:
Um. Okay - this is a bit odd (I do mention it in the docs - I invite you to tell me how I could express this isn the amount of text I can show in a menu) There are the 3 menu options:
Default - Printf is included in the binary **ONLY IF CALLED** and uses an intermediate amount of space. If you are not using printf(), you should never use any option other than default. 
Minimal - Printf is much smaller, but gets included whether you need it or not. (Why? Ask the avr-libc people, they admitted it was a limitation in the doc, they implied that it wasn't avoidabl) 
Full - Printf is much bigger, and gets included whether you need it or not. 

So if you see vprintf() overhead in the asm listing and memory map, that tells me that either an option (minimal from the sounds of it is selected, thereby forcing the compiler to include that version of vprintf, even though it may not be called. Or, of course, you're actually are calling it, and causing it to be included that way 

On a 3227: 

Empty sketch, 326 bytes. 
Empty sketch, printf set to minimal 1298 bytes
Empty sketch, printf set to full 3110 bytes

Okay, let's open serial, and print something to it with printf, and check the size. F
Printing millis with 100ms delay between them
Full 4661
Minimal 2849
Default 3221

Now, the format string was just "Millis: %d"

So I replaced with Serial.print("Millis: "); Serial.println(millis());

Full: 4704
Minimal: 2888
Default: 1916, because now that you're not forcing it to include a certain version of the library, it's free to omit it if it can never be called.

BTW, Default with the other overhead of the test sketch (Serial and millis()), that got me to 
1490 bytes, so Serial.print is just over 400, which is expected. Think about what you have to do to print a decimal number!
```c++
// Note: Compiled with millis disabled, hence the use of the util-delay, which is a tiny cycle-counting busywait. 
// default printf();
#include <util/delay.h>
void setup() {
  Serial.begin(115200);
  
}

void loop() {
  _delay_ms(100);
  Serial.print("hah "); 
  Serial.println("1000");
}
```
1459 bytes

Remove quotes around "1000":

1596 bytes

Replace with 
Serial.printHexln(1000); // prints 03E8

1492 bytes

Replace with 

Serial.printf("hah %d",1000);

2893

closed 2025-05-07 by SpenceKonde

Avamander, 2025-05-11:
> Default - Printf is included in the binary ONLY IF CALLED and uses an intermediate amount of space. [...] Or, of course, you're actually are calling it, and causing it to be included that way

Hmm, so `Serial` is using it in the background even if I only use `write` most likely? That's a bit unfortunate, but thanks for the clarification.

hmeijdam, 2025-05-11:
You could try to use the UART directly.

I use this approach if I am short on space and want to print some debug info.

Compilation result:  `Sketch uses 528 bytes (6%) of program storage space. Maximum is 7680 bytes.`

```
#include <util/delay.h>      //saves memory compared to standard delay()
char printbuffer[7];         //the ASCII of the integer will be stored in this char array
const static uint8_t Sometext[] = " and counting !\r\n"; // A string array with some fixed text
int8_t a; // this value will be used 

//#define USE_ALTERNATE_TXD  //  uncomment if you want to use the alternate TXD* / RXD* pins

void setup() {
  USART_init(9600);    //Call the USART initialization code and choose your ( baudrate )
}

void loop() {
  a = a + 1; // increasing value that will be printed
  itoa(a, printbuffer, 10);    //(integer, yourBuffer, base)
  USART_putstring(printbuffer); // print the
  USART_putstring(Sometext);
  _delay_ms(1000);          // saves about 30 bytes compared to standard delay()
//delay(500); //
}

void USART_init(uint32_t BAUDRATE) {
  USART0_BAUD = (F_CPU * 4 / BAUDRATE);
#ifdef USE_ALTERNATE_TXD
  PORTMUX_CTRLB = PORTMUX_USART0_bm; // activate alternate RXD* / TXD* pins
  VPORTA_DIR |= PIN1_bm;   // set pin PA1 as output
#else
  VPORTB_DIR |= PIN2_bm;   // set pin PA6 as output
#endif
  USART0_CTRLB |= USART_TXEN_bm ; //enable tx only
}

void USART_send(unsigned char data) {
  while (!(USART0_STATUS & USART_DREIF_bm));
  USART0_TXDATAL = data;
}

void USART_putstring(char* StringPtr) {
  while (*StringPtr != 0x00) {
    USART_send(*StringPtr);
    StringPtr++;
  }
}
```