SpenceKonde/megaTinyCore #1228: Unclear how bus arbitration errors should be handled

Sojourneer, 2025-05-05:
In the Wire library's two_master_write example, it is not clear how the endTransmission arbitration error (return 0x12) should be handled.  Is it sufficient to replay some portion or all of the API calls on the sending master, and if so which calls?  Does the client need to do something?

Also, is it confirmed that business arbitration errors are reliably detected in all cases?

MX682X, 2025-05-23:
Basically, bus arbitration means another master has taken over the bus a the same time (technically another slave could also do that, but only if they have the same address, so ignoring this). In general, this error means the complete message was lost and has to be resent.

SpenceKonde, 2025-05-24:
If any of you would like to write that up better and PR it into the readme for that?

SpenceKonde, 2025-05-31:
I definitely don't think we should make people reissue the entire startTransmission(), write() sequence over again - all those are doing is filling buffers. I'm not really following the code as I read it now, so I'm not sure quite how to make the change, but since we haven't  don'e anything to those buffers yet, we should just be able to not clear them right? And if they didn't resend, and instead tried to do something else with I2C, 

Certainly for the case of the lost arbitration, wherein we're almost guaranteed that just reattempting the transaction will work. 

On the other hand...
All the timeout errors are errors that generally indicate the either a software defect that must be corrected (like incorrect address) or a hardware problem, as do line state errors. Really, the **only** case that users ought to be retrying failed I2C transactions is if they got an arbitration lost.

I'm not sure what, if anything, gives "Unknown Error", or whether that is something that might work if retried.

closed 2025-06-01 by SpenceKonde