SpenceKonde/megaTinyCore #1238: Is the PTC gain default value PTC_GAIN_1 ?  and a easy typo issue

zbn10, 2025-07-02:
I have tried to use megaavr/libraries/PTC and i found a place where the Gain values seem to be insonsistent.

In ptc_node_set_gain(),  the argument 'gain' is subtracted from PTC_GAIN_MAX and shifted left.
`uint8_t ptc_node_set_gain(cap_sensor_t *node, ptc_gain_t gain) {
  gain = PTC_GAIN_MAX - gain;
  gain <<= 4;
  uint8_t ovs = node->hw_gain_ovs & 0x0F;
  node->hw_gain_ovs = gain | ovs;`

But in ptc_add_node(), the 'gain' default value is set to 0.
`node->hw_gain_ovs = NODE_GAIN(0, ADC_SAMPNUM_ACC16_gc);
#define NODE_GAIN(_a_, _d_)      (uint8_t)(((_a_) << 4) | ((_d_) & 0x0F))`

Is the PTC gain default value PTC_GAIN_1 ? 

`typedef enum ptc_gain_enum {
  PTC_GAIN_1            = 0x00,
  PTC_GAIN_2            = 0x23,
  PTC_GAIN_4            = 0x34,
  PTC_GAIN_8            = 0x3A,
  PTC_GAIN_16           = 0x3C,
  PTC_GAIN_32           = 0x3E,
  PTC_GAIN_MAX          = 0x3F,
} ptc_gain_t;`

SpenceKonde, 2025-07-13:
Sorry, I didn't write this library, so I don't know what this means to do.... @MX682X was the one behind the PTC libraries.

zbn10, 2025-07-16:
thanks.
I’ll try to ask on the MX682X repository.

closed 2025-07-16 by zbn10

MX682X, 2025-07-16:
Yeah... that's a bug and a typo.
When you work with some hand decoded assembly but try to optimize the code in a different way mix ups will happen. It really doesn't help that the higher the value, the lower the gain (probably because it directly affects a capacitance) and the orignal uses a Lookup-table.

I have to prepare an update anyway, to sync up the three differnt versions.
Will open a PR soon, when I have tested everything.