SpenceKonde/megaTinyCore #1245: Error on update to 2.6.11 in Arduino

CreativeRobotics, 2025-09-04:
I got this error when trying to update to yesterdays 2.6.11 release:

> Failed to install platform: 'megaTinyCore:megaavr:2.6.11'.
> Error: 13 INTERNAL: Cannot install tool DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino8: extracting archive: Open file C:\Users\bbigg\AppData\Local\Arduino15\tmp\package-967925964\avr-gcc-7.3.0-atmel3.6.1-azduino8-i686-w64-mingw32.tar: zip: unsupported compression algorithm

Nicks7777, 2025-09-05:
Same here.

I am using Arduino IDE 2.1.1 for Windows.

hmeijdam, 2025-09-05:
2.6.10 >> 2.6.11 board manager update on IDE 1.8.19 portable fails with the following error:

```
 Invalid archive: it must contain a single root folder
java.lang.RuntimeException: java.io.IOException: Invalid archive: it must contain a single root folder
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Invalid archive: it must contain a single root folder
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:170)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:82)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:141)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more

```

webtronico, 2025-09-05:
I am using Arduino IDE 2.3.7-nightly-20250830  for Windows.
Il tool DxCore:avrdude@6.3.0-arduino17or18 è già installato
Il tool arduino:arduinoOTA@1.3.0 è già installato
Il tool megaTinyCore:python3@3.7.2-post1 è già installato
Sto scaricando i pacchetti
DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino8
megaTinyCore:megaavr@2.6.11
Installazione DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino8
Failed to install platform: 'megaTinyCore:megaavr:2.6.11'.
Error: 13 INTERNAL: Non è possibile installare il tool DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino8: estrazione dell'archivio: Open file C:\Users\webtr\AppData\Local\Arduino15\tmp\package-355282005\avr-gcc-7.3.0-atmel3.6.1-azduino8-i686-w64-mingw32.tar: zip: unsupported compression algorithm

underwoodblog, 2025-09-07:
same here, ide 1.8.19

[image]

SpenceKonde, 2025-09-07:
Also, wtf, I was checking the issues every day for the past 4 days and didn't see this issue?!

cvmanjoo, 2025-09-07:
Happens and its okay😊

HenkHoldijk, 2025-09-07:
Murphy ... These things happen.

webtronico, 2025-09-07:
I tried installing it again now, but unfortunately I get this error:

The tool DxCore:avrdude@6.3.0-arduino17or18 is already installed
The tool arduino:arduinoOTA@1.3.0 is already installed
The tool megaTinyCore:python3@3.7.2-post1 is already installed
I'm downloading the packages
DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino8
Failed to install platform: 'megaTinyCore:megaavr:2.6.11'.
Error: 13 INTERNAL: The server responded with: 404 Not Found

SpenceKonde, 2025-09-08:
should be working now

closed 2025-09-08 by SpenceKonde

hmeijdam, 2025-09-08:
The installation works fine now, however I think with the change in toolchain something might have been missed.
note: When launching board manager Megatinycore seemed no longer installen. Installing the last version took only a few seconds. So I suspected that something was off. I removed megatinycore via boards manager. Deleted all the package and staging files to have a clean boards manager install. That worked as expected (the download took some time), however now I get this error about a missing tool (IDE 1.8.19 portable)

exec: "F:\\Arduino\\arduino-1.8.19\\portable\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino8/bin/avr-g++": file does not exist

full error window copy:
```
Arduino: 1.8.19 (Windows 10), Board: "ATtiny412/402/212/202, ATtiny412, 20 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), PA1-3,7 (default), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

F:\Arduino\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware F:\Arduino\arduino-1.8.19\hardware -hardware F:\Arduino\arduino-1.8.19\portable\packages -tools F:\Arduino\arduino-1.8.19\tools-builder -tools F:\Arduino\arduino-1.8.19\hardware\tools\avr -tools F:\Arduino\arduino-1.8.19\portable\packages -built-in-libraries F:\Arduino\arduino-1.8.19\libraries -libraries F:\Arduino\arduino-1.8.19\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=412,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,startuptime=8,wiremode=mors,printf=default,PWMmux=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=0000_0000 -ide-version=10819 -build-path C:\Users\hmeij\AppData\Local\Temp\arduino_build_52732 -warnings=all -build-cache C:\Users\hmeij\AppData\Local\Temp\arduino_cache_292550 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino8 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino8.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino8 -prefs=runtime.tools.python3.path=F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=F:\Arduino\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=F:\Arduino\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose F:\Arduino\Sketches\Attiny412_TCD0_input_PWM_capture_on_events_V3\Attiny412_TCD0_input_PWM_capture_on_events_V3.ino

F:\Arduino\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware F:\Arduino\arduino-1.8.19\hardware -hardware F:\Arduino\arduino-1.8.19\portable\packages -tools F:\Arduino\arduino-1.8.19\tools-builder -tools F:\Arduino\arduino-1.8.19\hardware\tools\avr -tools F:\Arduino\arduino-1.8.19\portable\packages -built-in-libraries F:\Arduino\arduino-1.8.19\libraries -libraries F:\Arduino\arduino-1.8.19\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=412,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,startuptime=8,wiremode=mors,printf=default,PWMmux=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=0000_0000 -ide-version=10819 -build-path C:\Users\hmeij\AppData\Local\Temp\arduino_build_52732 -warnings=all -build-cache C:\Users\hmeij\AppData\Local\Temp\arduino_cache_292550 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino8 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino8.path=F:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino8 -prefs=runtime.tools.python3.path=F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=F:\Arduino\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=F:\Arduino\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose F:\Arduino\Sketches\Attiny412_TCD0_input_PWM_capture_on_events_V3\Attiny412_TCD0_input_PWM_capture_on_events_V3.ino

Using board 'atxy2' from platform in folder: F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\hardware\megaavr\2.6.11

Using core 'megatinycore' from platform in folder: F:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\hardware\megaavr\2.6.11

Detecting libraries used...

[... cut ...] \\megaTinyCore\\hardware\\megaavr\\2.6.11\\cores\\megatinycore" "-IF:\\Arduino\\arduino-1.8.19\\portable\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.11\\variants\\txy2" "C:\\Users\\hmeij\\AppData\\Local\\Temp\\arduino_build_52732\\sketch\\Attiny412_TCD0_input_PWM_capture_on_events_V3.ino.cpp" -o nul

exec: "F:\\Arduino\\arduino-1.8.19\\portable\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino8/bin/avr-g++": file does not exist

Error compiling for board ATtiny412/402/212/202.

```